pauldemarco/flutter_blue #19: CancelConnect might have same issue as stopScan

brianegan, 2017-09-05:
Hey hey :) Just noticed `connect` and `cancelConnection`... that looks to have the same issue as the other PR we merged with `startScan` / `stopScan`

pauldemarco, 2017-09-05:
Although the context might be similar, this needs further consideration.
Neither iOS or Android time-out their connection requests (unless autoConnect = false on Android).
This means the user must keep a handle to a connectionSubscription for as long as they would _like_ to be connected to the device, not necessarily meaning they _are_ connected.
It may require an android edit, but at the moment I'm not seeing any reason why this wouldn't work.

pauldemarco, 2017-09-06:
What to return from the Stream is up for debate, I figured a BluetoothDeviceState would make the most sense.  With this, the listener can kick off discoverServices once the device has connected.
Any thoughts?

brianegan, 2017-09-06:
Ah yah, I see the dilemma...

My original thought wasn't to convert the Future to a Stream, as Futures deliver a Single Value (which is what you have in this case). Rather, my idea was to add a `whenComplete(closeConnection)` to the end of your Future chain that would ensure the methodChannel was closed automagically.

However, given the connection must remain open while using the device, I'd keep returning a Future. However, the `closeConnection` might be a bit confusing...

What you could do instead is add a `disconnect()`-type method to the `BlueToothDevice` instead. Then the api would look like:

```dart
final device = await flutterBlue.connect(someDeviceId)

device.doSomeStuff();

device.disconnect();
```

sandangel, 2018-08-18:
With current API and flutter hot reload feature, sometimes I found that the reference to the connection often gets lost between hot reload. And if the auto reconnect function is turn on, there is no way to disconnect except shutting down the app. The API that @brianegan proposed looks easier to use for me.

wkornewald, 2018-09-12:
@pauldemarco Is there a reason why you set `autoConnect` to `true`? The problem is that this makes `connect()` take a very long time (often more than 30s) and our device turns itself off before the connection can be established. With `autoConnect` set to `false` the connection is established almost instantly.