pauldemarco/flutter_blue #20: iOS support

pauldemarco, 2017-09-06:
An Objective-C implementation is still needed on iOS side.

JohannesTr, 2017-12-12:
Hi, is the iOS implementation in development at the moment?

pauldemarco, 2017-12-12:
Hi Johannes,

It is not, feel free to send PRs :)

HelenGuov, 2018-01-25:
Hi @pauldemarco I'm planning to do a PR for this issue. I'm trying to to use CoreBluetooth Objective C framework to implement the native functionality, but I can't seem to figure out a way to install the new framework into ios Flutter. Do you have any resources for this? 
Thanks

zerox1212, 2018-03-01:
Is there a branch for this somewhere? Is it possible to just port cordova iOS plugin? 
https://github.com/don/cordova-plugin-ble-central

pauldemarco, 2018-03-01:
I designed the API with both Android and iOS in mind, it should be fairly straight forward to implement.  I unfortunately don't have the time or the Objective-C experience to get this done at the moment.

Feel free to send PR's :+1:

zerox1212, 2018-03-01:
I don't know Objective-C either, but the Cordova plugin has a very similar API and is Apache 2.0 licensed. If someone could take a look and see if this could speed up implementation it would be great.

zerox1212, 2018-03-01:
I looked into it more and I think if someone can implement just one of the methods (such as `scan`) I could figure out the rest.

pauldemarco, 2018-03-07:
The majority of functionality has been implemented for iOS, with the exception of a couple minor issues needing Dart and Android changes as well.

I'll be closing this, please open up new issues if you run into iOS bugs.

Thanks everyone for your patience!

closed 2018-03-07 by pauldemarco