pauldemarco/flutter_blue #24: NPE thrown from Android Runtime after initiating scan in the example

samjxn, 2017-09-12:
Attempting to run the example code on a Nexus 7 (running android 6.0.1)

I'm consistently getting this stacktrace and an application crash a few seconds after initiating a scan:

```
E/AndroidRuntime( 7688): FATAL EXCEPTION: main
E/AndroidRuntime( 7688): Process: com.pauldemarco.flutterblueexample, PID: 7688
E/AndroidRuntime( 7688): java.lang.NullPointerException
E/AndroidRuntime( 7688): 	at com.pauldemarco.flutterblue.Protos$BluetoothDevice.setName(Protos.java:2850)
E/AndroidRuntime( 7688): 	at com.pauldemarco.flutterblue.Protos$BluetoothDevice.access$4900(Protos.java:2689)
E/AndroidRuntime( 7688): 	at com.pauldemarco.flutterblue.Protos$BluetoothDevice$Builder.setName(Protos.java:3089)
E/AndroidRuntime( 7688): 	at com.pauldemarco.flutterblue.ProtoMaker.from(ProtoMaker.java:38)
E/AndroidRuntime( 7688): 	at com.pauldemarco.flutterblue.ProtoMaker.from(ProtoMaker.java:28)
E/AndroidRuntime( 7688): 	at com.pauldemarco.flutterblue.FlutterBluePlugin$2.onScanResult(FlutterBluePlugin.java:579)
E/AndroidRuntime( 7688): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:355)
E/AndroidRuntime( 7688): 	at android.os.Handler.handleCallback(Handler.java:739)
E/AndroidRuntime( 7688): 	at android.os.Handler.dispatchMessage(Handler.java:95)
E/AndroidRuntime( 7688): 	at android.os.Looper.loop(Looper.java:148)
E/AndroidRuntime( 7688): 	at android.app.ActivityThread.main(ActivityThread.java:5417)
E/AndroidRuntime( 7688): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 7688): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
E/AndroidRuntime( 7688): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
```

pauldemarco, 2017-09-12:
It looks like android's BluetoothDevice.getName() can return a null value. From google:

> BluetoothDevice.getName() may return null if the name could not be determined. This could be due to any number of factors. Regardless, the name is the friendly name of the device, and shouldn't be used to distinguish it from other devices. 

This is causing line 38 of ProtoMaker.java to throw an NPE.

I think simply wrapping in a null check should suffice. Commit on the way...

pauldemarco, 2017-09-12:
Can you verify if that solves the issue?

samjxn, 2017-09-12:
Yep, that did the trick.  Thanks!

closed 2017-09-12 by samjxn