pauldemarco/flutter_blue #30: bluetoothDevice.discoverServices() doesn't return anything

dturing, 2018-01-30:
I'm having trouble getting the list of services from a bt device, my snippet:

```
List<BluetoothService> services = await widget.device.discoverServices();
print("success discovered");
```
never gets to print "success", here's what i see in the log:
```
D/FlutterBluePlugin(24804): onServicesDiscovered: 4 sink:null
```

It seems the servicesDiscoveredHandler.onListen function (which registers the servicesDiscoveredSink) is called only after onServicesDiscovered; but i can't find out why.

Maybe some change in the flutter framework? I've tried using alpha channel (2017-12-12) and git from yesterday. This seems to happen with flutter_blue 0.2.3 and git HEAD.

If you can't find time to fix this, any hints would be appreciated to maybe enable me to produce a PR.

dturing, 2018-01-31:
So my proposed patch seemed to fix the issue, but it's not all reliable. Before, the "sink:null" issue would occur most of the time; after the patch, it occurs much more rarely, but still does. I'll keep investigating, but am grateful for any help.

dturing, 2018-02-01:
I now think the problem is way deeper than just this.

It seems [all Bluetooth operations should be queued](https://stackoverflow.com/questions/39532768/queue-android-gatt-operations) as to avoid the otherwise common "PlatformException(read_characteristic_error, unknown reason, may occur if readCharacteristic was called before last read finished.".

If i want to read a number of characteristics, i have to await each of them before submitting the next. While this can work (although i'd want to dispute putting the "queueing burden" on the flutter_blue user), in the case of setNotification i don't think there's an easy way. There's no reply EventChannel set up, and in fact there are two operations (writing the CCCD and setting notify) that both return asynchronously from gattServer, so it's unclear how one could even wait for the setNotification to finish.

So to me it seems maybe the java FlutterBluePlugin should return results to those method calls only when they actually finish (their respective BluetoothGattCallback method is called), and also queue them up so only one such operation is attempted at any time? This way a user of flutter_blue wouldn't have to care about running only one method at the same time, and it might clean up the somewhat awkward setting up of extra response channels. It might also tune in well with #11.

Now, i'm not quite sure i can find the time to attempt this somewhat bigger operation. But if- @pauldemarco, do you agree with my rationale?

pauldemarco, 2018-04-03:
Hi dturing,
Sorry for the delay, I'd like to look into this soon, any updates?

dturing, 2018-04-03:
Cannot reproduce with current version from git, thank you.

closed 2018-04-03 by dturing

Mojachieee, 2018-08-17:
I'm still getting this error on the latest git version.

This will never print services:
    List<BluetoothService> services = await _device.discoverServices();
    print("Services");

The app also prints (if that's related)
`onServicesDiscovered: 5 sink:null`

I'm using Flutter Channel beta, v0.5.1

Let me know if you need any more details

Bakhoj, 2018-09-21:
I also have the problem with `sink:null`. 
My solution for now is making the discoverService into a recursive method. 
I call the method when the state changes to `BluetoothDeviceState.connected` when subscribing to `BluetoothDevice.onStateChanged().listen()`.
```dart
BluetoothDevice device;
bool isServicesDiscovered = false;
List<BluetoothService> services = new List();

_discoverServices() {
  device.discoverServices().then((s) {
    services = s;
    _updateServices();
    isServicesDiscovered = true;
  });
  Future.delayed(Duration(milliseconds: 100), () {
    if(!isServicesDiscovered && device != null)
      _discoverServices();
  });
}
```

The `_discoverServices()` calls itself every 100ms until the services is found or the connection is no longer established (if you set `device = null` when you disconnect).
From the few test I've done it doesn't go further then till 2nd discoverServices before it is succesful.

ened, 2018-09-27:
This should be a lot simpler. I noticed it sometimes works fine, which may be related to how BLE service discovery works internally on a phone. My Samsung S5 may cache the list of discovered services, which then causes `onServicesDiscovered` to be called right away (e.g. before the `StreamHandlers` `onListen` method has been called).

My workaround looks like this:

``` dart
var services = await device
    .discoverServices()
    .timeout(Duration(seconds: 10), onTimeout: () => <BluetoothService>[]);

if (services.length == 0) {
  print("Service list empty, flutter_blue race condition?");
  services =
      await device.services.catchError((error) => <BluetoothService>[]);

  if (services.length == 0) {
    print("no services advertised!");
    _cleanupAndCancel();
    return;
  }
}
```

bjornwilliam, 2019-03-13:
I am experiencing an issue where services are discovered every time when I first connect to a bluetooth device. But if I turn the bluetooth device off and then on again, device.discoverServices() will never return.