pauldemarco/flutter_blue #33: Package protobuf has no versions that match >=0.5.5 <0.6.0 derived from:

MaartenKleijwegt, 2018-03-01:
When trying to add flutter_blue to the most basic app setup with no other dependencies in android studio 3.0  it gives this error message:

/home/maarten/flutter/bin/flutter --no-color packages get
Running "flutter packages get" in tanngrisnir...
Package protobuf has no versions that match >=0.5.5 <0.6.0 derived from:
- flutter_blue 0.2.4 depends on version ^0.5.5
pub get failed (1)
Process finished with exit code 1

this is my pubspec.yaml:

name: tanngrisnir
description: Remote control app for tanngrisnir

depend
dev_dependencies:
  flutter_test:
    sdk: flutter

# For information on the generic Dart part of this file, see the
# following page: https://www.dartlang.org/tools/pub/pubspec

# The following section is specific to Flutter.
flutter:

  # The following line ensures that the Material Icons font is
  # included with your application, so that you can use the icons in
  # the material Icons class.
  uses-material-design: trueencies:
  flutter_blue: "^0.2.4"
  flutter:
    sdk: flutter

  # The following adds the Cupertino Icons font to your application.
  # Use with the CupertinoIcons class for iOS style icons.
  cupertino_icons: ^0.1.0

dev_dependencies:
  flutter_test:
    sdk: flutter

# For information on the generic Dart part of this file, see the
# following page: https://www.dartlang.org/tools/pub/pubspec

# The following section is specific to Flutter.
flutter:

  # The following line ensures that the Material Icons font is
  # included with your application, so that you can use the icons in
  # the material Icons class.
  uses-material-design: true

Nudelsieb, 2018-03-02:
Same problem here in VSCode:
```
flutter packages get
Running "flutter packages get" in minit_prototype_flutter...
Package protobuf has no versions that match >=0.5.5 <0.6.0 derived from:
- flutter_blue 0.2.4 depends on version ^0.5.5
pub get failed (1)
exit code 1
```

Nudelsieb, 2018-03-02:
Partly solved it with this https://flutter.io/developing-packages/#conflict-resolution

```
dependencies:
...

  flutter_blue: ^0.2.4
  protobuf: ^0.6.0

...

dependency_overrides:
  protobuf: ^0.6.0
```
However, now it throws `MissingPluginException` for method `startScan`.

commit b30d6106 (2018-03-02): Updated protobuf dependency (fixes #33). Removed redundant dependencies in example's pubspec (fixes #3).
closed 2018-03-02 by pauldemarco