pauldemarco/flutter_blue #34: UUID Service Filter for Android and iOS

pauldemarco, 2018-03-05:
startScan should support the ability for filter discoverable devices by service UUIDs.

jack4git, 2018-03-17:
Aha -- I had assumed since the call flutterBlue.scan included the list of Guids that this was in place. This explains why filtering isn't happening.

I'll need this capability for my app to use flutter_blue ...

pauldemarco, 2018-04-03:
Closing in favor of #61 and #35

closed 2018-04-03 by pauldemarco

jack4git, 2018-04-03:
Paul, not sure why this was closed? Are you rolling this functionality into another open issue?

jack4git, 2018-04-03:
Thanks Paul. I was a bit confused because the Core Bluetooth doc you linked to there doesn’t relate to filtering scanned devices by service uuid.

pauldemarco, 2018-04-03:
You're right! I split into a separate issue #64 

Thanks for the heads up