pauldemarco/flutter_blue #36: iOS Connect options

pauldemarco, 2018-03-05:
iOS supports options while using connectPeripheral:
https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/peripheral_connection_options?language=objc

ride4sun, 2020-04-06:
Even the option right now in the API of connecting has no documentation. (Delay and Autoconnect)
In the code, it states it is Android only. In general, it would be nice to have a proper API documentation

pauldemarco, 2020-04-06:
Documentation is not near where I’d like it to be. The code will have to serve as the documentation until the API is completely baked.

ride4sun, 2020-04-07:
The public API should have at least basic documentation stating the purpose of the parameter. That literarily takes seconds per method and can be picked up when typing the code by the user of the API. I could not decipher it from the code what the parameter of `connect` are doing e.g, It probably takes more time to answer questions for you about the purpose of a method parameter. The same is valid for regression testing.

I will be frank, this is bad practice. There is always a reason for postponing it. Same for refactoring and regression testing.

pauldemarco, 2020-04-07:
A separate issue has been opened to track documentation:
https://github.com/pauldemarco/flutter_blue/issues/533