pauldemarco/flutter_blue #39: (iOS) Error when trying to write CCC descriptor

pauldemarco, 2018-03-07:
iOS will terminate if user tries to manually write the CCC descriptor:
'NSInternalInconsistencyException', 'Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:'

Handle this gracefully by blacklisting CCCD UUID before call is made.

coderLM, 2019-08-22:
How to "blacklisting CCCD UUID before call is made"，I'am an Android developer using flutter.

aytunch, 2019-10-31:
@pauldemarco  Hi,
What if we need to write to a descriptor in order to get heart rate from MiBand3? I am using flutter too, and I need to write to a descriptor. What should I do?

pauldemarco, 2019-10-31:
@aytunch This issue just refers to the CCC descriptor. Writing to other (writable) descriptors should work fine. Let me know if you run into issues.

aytunch, 2019-10-31:
@pauldemarco Ah ok, I will see if it CCC descriptor first then. Thanks for the quick response, take care

aytunch, 2019-10-31:
@pauldemarco I am still having issues, the moment I try to write to the descriptor, app crashes instantly. It should not be CCC but I could not be sure. Is there a way to ask the characteristic/descriptors of their traits (allow-read/write/notify)? I am trying to get Real Time Heart Rate from my MiBand3. My code and the result is this:

```
void _startMonitoringHeartRate() async {
    if (hrmctrlCharacteristic != null && hrmdataDescriptor != null) {
      print("NOT NULLLLLL");
      await hrmctrlCharacteristic.write(
          hex.decode("15" + COMMAND_SET_HR_MANUAL + COMMAND_STOP),
          withoutResponse: true);
      print("1111");
      await hrmctrlCharacteristic.write(
          hex.decode("15" + COMMAND_SET__HR_CONTINUOUS + COMMAND_STOP),
          withoutResponse: true);

      print("2222");
      print("descriptorUUID:${hrmdataDescriptor.uuid.toString()}");
      print("characterUUID:${hrmdataDescriptor.characteristicUuid.toString()}");
      print("serviceUUID:${hrmdataDescriptor.serviceUuid.toString()}");
      await hrmdataDescriptor.write(hex.decode("0100"));

      print("3333");
      await hrmctrlCharacteristic.write(
          hex.decode("15" + COMMAND_SET__HR_CONTINUOUS + COMMAND_START),
          withoutResponse: true); //TODO: change to true!!!!

      print("4444");
      Timer.periodic(Duration(seconds: 12), _sendPing(hrmctrlCharacteristic));

      print("5555");
    } else {
      print("NULLLLLL");
    }
  }
```

Output:

> flutter: Start Getting Continous Heart Data...
> flutter: NOT NULLLLLL
> flutter: 1111
> flutter: 2222
> flutter: descriptorUUID:00002902-0000-1000-8000-00805f9b34fb
> flutter: characterUUID:00002a37-0000-1000-8000-00805f9b34fb
> flutter: serviceUUID:0000180d-0000-1000-8000-00805f9b34fb
> Lost connection to device.
> Exited (sigterm)

One last thing: What is the _bool inside character.write(_data, withoutResponse:_bool) for? Is there a chance the problem might be caused of using wrong withoutResponse for the previous character.write's? And there is no `withoutResponse` inside of descriptor.write. What is the reason there?

Yellowstech, 2020-07-13:
Hello people!
I have the same problem!! I can't to write descriptor and i don't know what i do...
I'am using the flutter_blue 0.7.2

My code:
   targetCharacteristic = service.characteristics
        .where((element) => element.uuid.toString() == GLUCOSE_CHARACTERISTIC)
        .first;

    if (targetCharacteristic != null) {
      print("notify GLUCOSE_CHARACTERISTIC");
      targetCharacteristic.setNotifyValue(true);
      BluetoothDescriptor descriptor = targetCharacteristic.descriptors[0];
      if (descriptor != null) {
        List<int> elements = [1, 1];
        var data = new Uint16List.fromList(elements);
        await descriptor.write(data);
      }

      targetCharacteristic = null;
    }

Message error:
D/BluetoothGatt(24483): setCharacteristicNotification() - uuid: 00002a18-0000-1000-8000-00805f9b34fb enable: true
E/flutter (24483): [ERROR:flutter/lib/ui/ui_dart_state.cc(166)] Unhandled Exception: PlatformException(RuntimeException, Protocol message contained an invalid tag (zero)., com.google.protobuf.InvalidProtocolBufferException: Protocol message contained an invalid tag (zero).
E/flutter (24483):      at com.google.protobuf.CodedInputStream.readTag(CodedInputStream.java:148)

woopluz, 2020-09-23:
> How to "blacklisting CCCD UUID before call is made"，I'am an Android developer using flutter
I hava got the same issue ,do you deal it ? @coderLM @pauldemarco

claytonsands, 2020-10-02:
Hello people!
Has anyone here managed to solve the bluetooth communication problem with the miband?

I was able to decipher the authentication process with the bracelet, but I'm having trouble implementing it in flutter. Can you help me?

The steps are as follows:
1- Create a key with 16bytes. Example: {0x30, 0x31, ..., 0x46, 0x45}
2 - Enable notifications on the miband by sending TRUE to service UUID 0x0009
3 - Send {**0x01**, 0x08, KEY} to UUID 0x0009.
4 - Check if the first three bytes of the return were {0x10, **0x01**, 0x01} which means success in receiving.
5 - Send another request now with an extra bit in the first element and without a key {**0x02**, 0x08}
6 - Check the return again if equal to {0x10, **0x02**, 0x01} means success.
7 - Take the last 16bytes of the response from step 6 and encrypt it using AES / ECB / NoPadding using the key created in step one. Store in a RESPONSE_KEY variable.
8- Send another request with the encrypted response created above. Example: {**0x03**, 0x08, RESPONSE_KEY}, similar to what is done in step 3.
9- Check the return, if equal to {0x10, **0x03**, 0x01} the data has probably already been sent.

Font: https://leojrfs.github.io/writing/miband2-part1-auth/