pauldemarco/flutter_blue #40: (iOS) Flutter run sometimes errors

pauldemarco, 2018-03-07:
This error is sometimes experienced when trying to `flutter run`:
```
[!] ERROR: Parsing unable to continue due to parsing error:
    contained in the file located at /Users/pauldemarco/flutter_blue/example/ios/Podfile.lock
    PODS:
      - !ProtoCompiler (3.4.0):
        - Protobuf (~> 3.0)
      - Flutter (1.0.0)
      - flutter_blue (0.0.1):
        - !ProtoCompiler
        - Flutter
        - flutter_blue/Protos (= 0.0.1)
      - flutter_blue/Protos (0.0.1):
        - !ProtoCompiler
        - Flutter
        - Protobuf
      - Protobuf (3.5.0)
    
    DEPENDENCIES:
      - Flutter (from `/Users/pauldemarco/flutter/bin/cache/artifacts/engine/ios`)
      - flutter_blue (from `/Users/pauldemarco/flutter_blue/ios`)
    
    EXTERNAL SOURCES:
      Flutter:
        :path: /Users/pauldemarco/flutter/bin/cache/artifacts/engine/ios
      flutter_blue:
        :path: /Users/pauldemarco/flutter_blue/ios
    
    SPEC CHECKSUMS:
      !ProtoCompiler: 07d0c441bc00e7f01e84debf7c53794683fbca7c
      Flutter: 58dd7d1b27887414a370fcccb9e645c08ffd7a6a
      flutter_blue: 6a29a637fac55c8918a01e9d2c3d381612b1357a
      Protobuf: 8a9838fba8dae3389230e1b7f8c104aa32389c03
    
    PODFILE CHECKSUM: 351e02e34b831289961ec3558a535cbd2c4965d2
    
    COCOAPODS: 1.3.1
    
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-core-1.2.1/lib/cocoapods-core/yaml_helper.rb:62:in `rescue in load_string'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-core-1.2.1/lib/cocoapods-core/yaml_helper.rb:57:in `load_string'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-core-1.2.1/lib/cocoapods-core/yaml_helper.rb:75:in `load_file'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-core-1.2.1/lib/cocoapods-core/lockfile.rb:40:in `from_file'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.2.1/lib/cocoapods/config.rb:199:in `lockfile'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.2.1/lib/cocoapods/command.rb:138:in `installer_for_config'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.2.1/lib/cocoapods/command/install.rb:38:in `run'
    /Library/Ruby/Gems/2.3.0/gems/claide-1.0.2/lib/claide/command.rb:334:in `run'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.2.1/lib/cocoapods/command.rb:52:in `run'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.2.1/bin/pod:55:in `<top (required)>'
    /usr/local/bin/pod:22:in `load'
    /usr/local/bin/pod:22:in `<main>'
Error running pod install
```

pauldemarco, 2018-03-07:
Solution consisted of the following:
1. Install cocoapods 1.2.1
`sudo gem uninstall cocoapods`
`sudo gem uninstall cocoapods-core`
`sudo gem uninstall cocoapods-deintegrate`
`sudo gem uninstall cocoapods-downloader`
`sudo gem uninstall -aIx`
`sudo gem install -n /usr/local/bin cocoapods -v 1.2.1`

2. Reinstall pods
`cd ./flutter_blue/example/ios`
`rm -rf ./Podspec.lock`
`export FLUTTER_FRAMEWORK_DIR=/Users/pauldemarco/flutter/bin/cache/artifacts/engine/ios`
`pod install`

Now, `flutter run` should run successfully.

closed 2018-03-07 by pauldemarco

Albinzr, 2018-04-03:
[image]
@pauldemarco  This is the error I'm getting, Can you tell me whats happening?

Albinzr, 2018-04-03:
@pauldemarco  after i did the fix you said I'm getting new set of errors
[image]

psycura, 2018-12-10:
Did all the steps from above - but still receive error:

> Error output from CocoaPods:
> ↳
>        /bin/bash: line 3: /Users/eugenealitz/projects/ble_test/Pods/!ProtoCompiler/protoc: No such file or directory
> 
>     [!] Automatically assigning platform ios with version 8.0 on target Runner because no platform was specified. Please specify a platform for this target in your Podfile. See `https://guides.cocoapods.org/syntax/podfile.html#platform`.

tommaso1, 2019-04-19:
I found and easier solution, just
`cd ios`
`pod install`
`cd ..`
`flutter run`

TheWalkingDead1024, 2019-09-30:
still stuck in this issue. still have no idea.

danfleck, 2019-10-10:
Any ideas on what to do for this? I fixed the protoc problem and I'm still getting errors. Running using Flutter Run in Android Studio:
`/flutter_blue/ios/Classes/FlutterBluePlugin.h' is located outside of the allowed root paths.`

Using XCode: 
'flutter_blue/FlutterBluePlugin.h' file not found'

Any ideas on debugging or what to try to get ios working?