pauldemarco/flutter_blue #44: Only one characteristic can notify at a time

janekm, 2018-03-09:
Unless I am mistaken, the following code only allows one notification to be active at a time:
```
/// Notifies when the characteristic's value has changed.
  /// setNotification() should be run first to enable them on the peripheral
  Stream<List<int>> onValueChanged(BluetoothCharacteristic characteristic) {
    return FlutterBlue.instance._characteristicNotifiedChannel
        .receiveBroadcastStream()
        .map((buffer) => new protos.OnNotificationResponse.fromBuffer(buffer))
        .where((p) => p.remoteId == id.toString())
        .map((p) => new BluetoothCharacteristic.fromProto(p.characteristic))
        .where((c) => c.uuid == characteristic.uuid)
        .map((c) {
          characteristic.updateDescriptors(c.descriptors);
          characteristic.value = c.value;
          return c.value;
        });
  }

```
For many BLE devices it is necessary to listen to notifications from multiple characteristics at the same time. Not to mention from multiple devices!

pauldemarco, 2018-03-09:
This allows for multiple characteristics, since they are being filtered first by device ID and then by characteristic uuid.

Have you experienced an issue with this? If so, I can investigate.

pauldemarco, 2018-03-09:
It looks like .receiveBroadcastStream() will override the previous platform channel if it's called more than once.  Good catch!

I'll use this opportunity to refactor some event channels to utilize the main method channel (#11).

Fix on the way.

closed 2018-03-09 by pauldemarco

janekm, 2018-03-10:
That was quick! I can confirm it works for me now. Thanks :)

commit 09eb37ca (2018-03-09): OnValueChanged moved to method channel (#11),fixes #44. iOS discoverServices queue implemented, fixes #46.