pauldemarco/flutter_blue #45: Standardize logging across entire plugin

pauldemarco, 2018-03-09:
Platform code contains logging that should be user-configurable.

Allow the user to change verbosity level for the plugin.

pauldemarco, 2018-12-14:
Model after [journald](https://www.freedesktop.org/software/systemd/man/journald.conf.html):

> ... takes one of "emerg", "alert", "crit", "err", "warning", "notice", "info", "debug", or integer values in the range of 0–7 (corresponding to the same levels). Messages equal or below the log level specified are stored/forwarded, messages above are dropped.

```dart
enum LogLevel {
    emergency,
    alert,
    critical,
    error,
    warning,
    notice,
    info,
    debug,
}
```
Allow LogLevel to be set at any time:
```dart
FlutterBlue.instance.setLogLevel(LogLevel level);
```
LogLevel must also be passed to platform side to be used by underlying Android/iOS.

pauldemarco, 2019-09-12:
Emulate the format of error messages in latest Flutter (1.9)

pauldemarco, 2019-12-10:
Log messages need to be changed from `NSLog` to `log` on iOS.

maks, 2021-02-08:
@pauldemarco Unfortunately this no longer seems to be woorking in `0.7.3` , after setting `FlutterBlue.instance.setLogLevel(LogLevel.critical);` still has all the notification debug messages being logged.

tylermcdonald, 2021-04-19:
Is this still an issue? Log level used to work for me, but I'm now running into the issues @maks mentioned. Any solutions or additions needed to fix this, or has it not been addressed yet?

lookacat, 2022-09-28:
any news on this? this package is awesome but the amount of logs make it hard to debug