pauldemarco/flutter_blue #47: Support for advertising

fvisticot, 2018-03-15:
It would be nice to support advertising as well as it is available on ios and Android( >=5.0)

fmatosqg, 2018-03-16:
I have 2 devices running the sample app and they can't see each other, I guess that would be the reason?

At least I think we should add it to the checklist on the README.md so people can be aware it's missing.

fmatosqg, 2018-03-16:
See https://github.com/pauldemarco/flutter_blue/commit/9b56d7cb3f80336111b062cd2aef24bdec4a7f0e for proposed readme change

fvisticot, 2018-03-16:
I quickly check the iOS plugin code and I do not see any code providing advertising support like this:

```
[myPeripheralManager startAdvertising:@{ CBAdvertisementDataServiceUUIDsKey :
--
@[myFirstService.UUID, mySecondService.UUID] }];
```

Is it planned to support advertising for Android and iOS ?

fmatosqg, 2018-03-19:
@fvisticot Android advertising is ready on commit #55 although iOS build is broken by a 3rd party dependency on permissions.

fmatosqg, 2018-04-18:
I'm really stuck and needing advertising. For short term I'll reference my fork commit on app project, but it would be nice to land it on main repo.

fvisticot, 2018-04-22:
any update for iOS advertising support ?

fmatosqg, 2018-04-22:
Hi @pauldemarco 
I'd respectfully consider a fork that supports advertising, since the PR is stuck waiting for your blessing for quite a while. 

I can solve the conflicts, but before I do that I would like to know if you're open to have a conversation to make it work.

As noted I need advertising on a project I'm working on, so it's hard for me to wait for long times for PR + publishing on pub.dart. If you're short on time we can make it work long term, but it's hard for me to wait without knowing what's going on.

cheers

cc @fvisticot

ravernkoh, 2018-04-29:
Hi @pauldemarco 
I also need advertising on a project I'm working on!

fmatosqg, 2018-04-30:
@ravernkoh on short term you can point your `flutter_blue` version to my fork. It doesn't have the latest commits from this repo, and I'm working on making it work with Dart 2.0, but it works for me on Android. 

If it gets traction I can put more time to merge with original repo master and perfectly compatible with Dart 2.0.
```
dependency_overrides:
  flutter_blue:
    git:
      url: https://github.com/fmatosqg/flutter_blue.git
      ref: issue_47_advertising
```

Gerriko, 2018-05-18:
Was about to make the same request (especially now that the physical web app was pulled by Google - a real pity as that app should have been a standard item on all phones, in my opinion).

Anyhow, This advertising feature, such as scanning for broadcasting only ble devices is sorely needed with support for all beacon formats and determining own formats and TLM type payloads etc. If it can be run in the background, even better. Please please let us know timelines. thanks.

TheBosZ, 2019-01-25:
@Gerriko I think you're misunderstanding this ticket.

I **believe** this ticket is asking for the ability to advertise **from** the phone, not read advertisements **from** beacons and devices.

Reading advertisements from beacons and the like is available in the scan results.