pauldemarco/flutter_blue #51: Dart 2 invokeMethod compatibility

xqwzts, 2018-03-17:
With the upcoming Dart 2 changes we will run into errors when returning anything other than `Future<dynamic>` from `invokeMethod` calls.

Calls like:
```dart
Future<bool> get isOn => _channel.invokeMethod('isOn');
```

Need to be updated to:
```dart
Future<bool> get isOn async {
 final bool isOn = await _channel.invokeMethod('isOn');
 return isOn;
}
```

More details in this issue: https://github.com/flutter/flutter/issues/15654

closed 2018-04-02 by pauldemarco

SUPERETDUPER, 2018-07-13:
Seems like this is not fully finished. The `flutterBlue.isAvailable` and `flutter.isOn` methods still throw errors

SUPERETDUPER, 2018-07-13:
I can submit a pull request but I need to know what is best practice (probably option 1).

**Option 1**
As @xqwzts recommands in his comment

**Option 2**
Simply change the return type from `Future<bool>` to `Future`

Thanks in advance!

itavero, 2018-10-08:
I think this issue needs to be reopened, as I am still getting errors when using `isOn` or `isAvailable` in v0.4.0.

SUPERETDUPER, 2018-10-08:
After more thought I think option when (from @xqwzts) is better since it doesn't change the method signature. I will submit a pull request when I have a moment.

commit 69493c9f (2018-04-02): Dart 2 compatibility. Fixes #51