pauldemarco/flutter_blue #54: AdvertisementData contains empty values on iOS

ryanhanks, 2018-03-19:
Hi,

On iOS, I'm not seeing any data for the following properties on `AdvertisementData` in my `ScanResult` event stream handler:
* `localName`
* `manufacturerData`
* `serviceData`

I observed this to be consistent over `ScanResult`s for responses for 19 different devices.

`localName` returns an empty `String`, and `manufacturerData` and `serviceData` return an empty `List<int>`.

Let me know if I can provide more details or if I can help in troubleshooting, debugging, or verifying a fix.

Thanks.

jack4git, 2018-03-23:
Confirming this behavior. The information is key, especially serviceData, otherwise you have to connect to devices to determine what services they support.

pauldemarco, 2018-04-03:
Looking into this now.

Not good news, as this is working on my end (using iPhone 8 plus and nrf52 dev board with nRF Connect for Desktop).

After setting up the advertisement data in nRF Connect, I'm seeing all of the values on Dart side.

Have you confirmed this information is showing up in other Bluetooth apps (nRF Connect for iOS)?

jack4git, 2018-04-03:
Paul, I’m also seeing some devices with advertisement data. Guessing it’s showing up if it’s there, but haven’t got around to changing the settings on one of my BLE devices yet to confirm.

jack4git, 2018-04-19:
Hey Paul,

I am now able to confirm that there's advertising packet data that I can't get to from flutter_blue. 

My BLE device is include a Service UUID in its advertising packet, which will allow my app to know that a given service is available BEFORE connecting. I'm guessing this should be coming through in ScanResult.advertisementData.serviceData ... 

I can see this data coming from my device using various apps such as Nordic's nRF Connect, where it looks like this:
[image]
When I access this and various other properties of AdvertisementData for my device in flutter_blue, all are empty. When I look at the same info for "all devices in range" I see just one device out of ~20 that shows a non-empty value for serviceData. I am able to see localName for my device. (I don't see other values but those aren't being sent). 
I can provide more info as needed.

jack4git, 2018-04-23:
Paul, following up on this one --

francisraj, 2018-07-02:
hi, 
   i'm trying to create an android and ios app in flutter for ble operations, i need to take all bytes of advertisement data while scanning time and i need to make a condition check based on the advertisement bytes. 
i can get all byte using NRF CONNECT (advertisement packet 1 and 2 total is 62 byte data), 
how to get from this?

[image]

could you please help me,..

hbsis-eduardo, 2018-07-12:
This error is also happening on Android. I'm using the example app to run on API 22.
I just downloaded the example and ran it.

pauldemarco, 2018-08-10:
@jack4git Could you share the "raw" table, like shown by @francisraj ?

francisraj, 2018-08-20:
> @jack4git <https://github.com/jack4git> Sorry for the delay. If you're
> still in need of a fix, please try out the latest version 0.4.0 on pub and
> let me know how it goes.
>
> Thanks!

Hi,
   Now we can get rssi and manufacturer data, but we need full bytes of
advertisement data ( it is  maybe 61 bytes data).

And how to make broadcastUpdate like android coding.

@Override
public void onCharacteristicRead(BluetoothGatt gatt,
                                 BluetoothGattCharacteristic characteristic,
                                 int status) {
    if (status == BluetoothGatt.GATT_SUCCESS) {

      *   broadcastUpdate(ACTION_DATA_AVAILABLE, characteristic);
//>>>>>>>>>>>>  Like this.*

    }
}

using this *broadcastupdate* i can easily find and filter to take correct
data from ble.

closed 2018-12-19 by ryanhanks

pauldemarco, 2018-12-19:
@ryanhanks Just confirming, is this issue fixed for you?

GauravPatni, 2019-02-08:
Paul,
I am also facing same issue . After scan Service data is null . ( flutter_blue , Master - 
Latest commit 6f6fa30 )

Thanks,
Gaurav

robomike, 2020-02-25:
Hi, I have the same issue.  using the latest build.  Any updates?

PeteSwad, 2020-05-08:
Hi

I have found that the ScanResult serviceData is overwritten by the devices scan response service data if it contains values.

Ideally ScanResult AdvertisingData would return both the ScanResult and ScanResponse service data in separate Lists. I guess this depends on how the OS's handle the Scan Response data though. 

I hope this helps someone

Pete

novas1r1, 2020-05-27:
Hi,
having the same issue but on Android. ServiceData and manufacturerData are empty for one of 3 different devices.
Using Pixel 2 with Android 10 and Flutter 1.17.1

recvec, 2020-07-31:
having the same issue but on Android 4.4.4, nRF Connect shows this data

Yinten, 2022-01-26:
Anyone find anything out on thiis?

AymenXD, 2023-02-24:
any fix guys? this is an example:

Android (works fine) :
[image]

iOS:
[image]
 same here :