pauldemarco/flutter_blue #57: Support concurrent characteristic reads via Future.wait

jack4git, 2018-03-27:
This code *should* allow concurrent reading of characteristics:
`
  Future.wait([
    device.readCharacteristic(char1),
    device.readCharacteristic(char2),
// etc.
  ])
  .then((returnsList) {
    print('In Future.wait.then, ${returnsList.length}');
  })
  .catchError((e) {
    print('error $e');
  });
`
But in fact it never gets to `.then` ... nor does it throw and error

Based on the debug messages coming out of flutter_blue it appears that both characteristics are indeed being read. So maybe there's a tweak needed to handle the case that multiple futures are active at the same time?

Covered here in Dart docs: [Async](https://www.dartlang.org/tutorials/language/futures)

pauldemarco, 2018-03-28:
I'll have some time to investigate this next week, thanks for reporting!

jack4git, 2018-03-28:
Tried this again after I learned some lessons about async ... another version of the code
```
Future.wait(_charValMap.keys.map(powerDevice.scanResult.device.readCharacteristic))
  .then((List<List<int>> listList) {
    print('Inside Future.wait, listList has ${listList.length} elements');
  })
  .catchError((e) {
    print('error $e');
  }); 
```

pauldemarco, 2018-03-28:
`Future.wait` will fire off the functions in _quick succession_. 

Both iOS and Android only support one readCharacteristic attempt at a time, or in other words, you must wait for the readCharacteristic to be return before calling it again.

Dart makes this easy with async functions like so:
```dart
readAll() async {
  final value1 = await device.readCharacteristic(char1);
  final value2 = await device.readCharacteristic(char2);
  // do something with value1 and value2
}
```

Here, the `await` keyword will wait for readCharacteristic to return before moving onto the next line of code.

If you wanted similar functionality to your function you could do something like:

```dart
Future<List> readAll() async {
  final value1 = await device.readCharacteristic(char1);
  final value2 = await device.readCharacteristic(char2);
  return [value1, value2];
}

readAll().then((values) => print('${values.length} read.'));
```

jack4git, 2018-03-28:
re: one attempt at a time ... I was not aware of the limitation in iOS / Android, can you point me at a reference?

re: firing off in quick succession, yes indeed. But based on what I'm seeing logged, it looks like flutter_blue is actually executing all the reads successfully ... the log looks just like it does on my successful approach using await in a loop ... I wonder if the right solution here is (assuming you can detect multiple overlapping read attempts) is to error out, or to actually allow it to work (sequentially) by returning futures for all requests when they complete ... right now it almost "succeeds silently" ...

Here's my current approach, given that flutter_blue gives me lists of characteristics: looping through a list (this just does the bluetooth value to internal model mapping on one characteristic at the moment):
```
for (BluetoothCharacteristic char in _charValMap.keys.toList()) {
    tmpIntList = await powerDevice.scanResult.device.readCharacteristic(char);
    _charValMap[char] = tmpIntList;
    print('processed _charValMap item: ${char.uuid} ${tmpIntList.toString()}');
    // Process/save values here ...
  }

pauldemarco, 2018-10-09:
I am unable to replicate this issue @jack4git 

Closing this for now, please reopen if you are still experiencing this in the latest version.

closed 2018-10-09 by pauldemarco

akshay-kapase, 2020-01-07:
There are three services out of which the third one should be used to read and write characteristic.The first two are generic and the third one is custom service using which it will work. Try this :

_scanResult.device.discoverServices().then((services){
        services[2].characteristics[0].read();
    });