pauldemarco/flutter_blue #59: Cannot run in iOS

Albinzr, 2018-04-02:
I have added flutter_blue to my yaml file after that I'm getting pod error

```
dependencies:
  flutter:
    sdk: flutter
  flutter_blue: "0.3.1"
  cupertino_icons: ^0.1.0
```
If I remove  flutter_blue then it starts to work normally
Why is this issue happening and how can I solve this?

pauldemarco, 2018-04-02:
Hi Albinzr,
What's the error message you're seeing?

Albinzr, 2018-04-03:
@pauldemarco After I did the fix you said, I'm getting new set of errors
[image]

pauldemarco, 2018-04-03:
Most of those are warnings but there is 1 error: `flutter_blue/FlutterBluePlugin.h file not found`...

Have you tried opening the example project in xcode?

Albinzr, 2018-04-03:
I created a new project and added it, then it started working fine.

closed 2018-04-03 by Albinzr

pauldemarco, 2018-04-03:
Thanks for the report!

truongsinh, 2019-03-10:
I have the same issue, and the culprit is this line
https://github.com/pauldemarco/flutter_blue/blob/ce43a748e949fa5b46fb97b0fac77d12d9238745/ios/flutter_blue.podspec#L21

Please be noticed that also in this line, the path will not work in Windows

mateuszjablonski, 2019-09-27:
Is this ever going to be fixed? I have exactly the same issue, and it is over a whole year since it first occured.

pauldemarco, 2019-09-27:
Multiple attempts have been made and reverted.  There doesn’t seem to be
one that works for everyone.  Leaving this open until there’s overwhelming
response, or until it’s addressed in an official flutter template.

mateuszjablonski, 2019-10-02:
I think I have managed to find a solution to the problem. It seems like setting headers as public solves the problem. [Detailed info here](https://dev.to/rkowase/how-to-fix-the-error-that-the-header-file-of-flutter-plugin-does-not-find-on-ios-build-4c22). @Albinzr @truongsinh can you confirm that this solves the problem?

KevinRohn, 2019-10-08:
I also have the same problem. Is there any update? Setting the headers as public works some times, but is not practicable for me, because i can't do this one codemagic.io which runs our builds. It only is possible on the local build machine.

activeRobert, 2019-10-10:
Any update on this?

davidcv5, 2019-10-11:
Anything new? Having this with firebase make it even worse, can't find a way to make it build, changing the path affects all firebase packages.

softmarshmallow, 2019-10-19:
> Anything new? Having this with firebase make it even worse, can't find a way to make it build, changing the path affects all firebase packages.

+1 for firebase with fluuter_blue integration. something's badly chained

jacobokoenig, 2019-11-20:
+1 also firebase with flutter_blue. It was working fine a while ago. Not sure what happened.