pauldemarco/flutter_blue #62: API for Serial communication

eugenelet, 2018-04-03:
Hi,

I would like to ask is there any way that I can use this plugin for serial communication with another bluetooth device? Is there anywhere that I can look into to implement this function?

Thanks.

pauldemarco, 2018-04-03:
Hi @eugenelet,

Are you referring to Bluetooth classic's serial port profile (SPP)?  If so, there really isn't a BLE equivalent.  Depending on your data rate needs, you could create a Bluetooth Low Energy service that offers similar functionality, which then could be handled by this library.

eugenelet, 2018-04-03:
Hi @pauldemarco 

Yeah that's what I need. Where can I start with that using this library? As from the functions listens in this plugin, there's readCharacteristic, services, readDescriptor, etc. I'm not sure which should I use to implement this function.

pauldemarco, 2018-04-03:
@eugenelet I would run through some basics of Bluetooth Low Energy, and see if an official service has what you need. 

In a nutshell, you could use this library to:
1. Scan for nearby devices.
1. Connect to device
1. Discover service on device
1. Transfer data by reading and writing to a service's characteristic

eugenelet, 2018-04-03:
@pauldemarco Thanks for helping out! It's been days for me looking around for a SPP solution using Flutter. So the basic idea is I've to use readCharacteristic and writeCharacteristic provided by this plugin to implement this function?

pauldemarco, 2018-04-03:
That's the idea 👍 

For each step you would need the following:
1. Scan for nearby devices. `FlutterBlue.instance.scan`
1. Connect to device. `FlutterBlue.instance.connect`
1. Discover services on device. `device.discoverServices`
1. Transfer data by reading and writing to a service's characteristic. `device.readCharacteristic device.writeCharacteristic`

eugenelet, 2018-04-03:
Thanks for the info. I'll try it out and let you know if it works. 😀

pauldemarco, 2018-04-03:
Sounds good. I'll close this for now, if you run into issues we can reopen.

closed 2018-04-03 by pauldemarco

faridishere, 2018-08-08:
Hi 
I am running into a similar issue. I am trying to connect to my laptop. The example code can find the laptop (although it does not show the assigned name and just shows blank.)

When I try to connect it does not even when I increased the timeout.

Thanks for the help.

ayushin, 2020-04-29:
Anybody else is working on this?