pauldemarco/flutter_blue #66: ArrayIndexOutOfBoundsException from Android Runtime when scan is initiated

contango1, 2018-04-06:
Running example app on Android 6.0, I get this exception when scan is initiated.

```
D/BluetoothDevice(11526): mAddress: 25:0A:C4:0C:FF:2C
D/BluetoothDevice(11526): getName: name = MyBLE
D/BluetoothDevice(11526): getType: type = 2
D/AndroidRuntime(11526): Shutting down VM
E/AndroidRuntime(11526): FATAL EXCEPTION: main
E/AndroidRuntime(11526): Process: com.pauldemarco.flutterblueexample, PID: 11526
E/AndroidRuntime(11526): java.lang.ArrayIndexOutOfBoundsException: Not enough data.
E/AndroidRuntime(11526):        at com.pauldemarco.flutterblue.AdvertisementParser.parse(AdvertisementParser.java:60)
E/AndroidRuntime(11526):        at com.pauldemarco.flutterblue.ProtoMaker.from(ProtoMaker.java:30)
E/AndroidRuntime(11526):        at com.pauldemarco.flutterblue.FlutterBluePlugin$2.onScanResult(FlutterBluePlugin.java:625)
E/AndroidRuntime(11526):        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:362)
E/AndroidRuntime(11526):        at android.os.Handler.handleCallback(Handler.java:815)
E/AndroidRuntime(11526):        at android.os.Handler.dispatchMessage(Handler.java:104)
E/AndroidRuntime(11526):        at android.os.Looper.loop(Looper.java:207)
E/AndroidRuntime(11526):        at android.app.ActivityThread.main(ActivityThread.java:5728)
E/AndroidRuntime(11526):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(11526):        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:789)
E/AndroidRuntime(11526):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:679)
```

MSchrenker, 2018-04-13:
I got the same issue, with a specific hardware device in development. There are two times a long local name (Type 9) in it and no Type 8 part (for the short local name). The implementation of the AdvertismentParser breaks out of the loop as soon as it founds a second long local name, without moving the position in the bytestream to parse further. 

I tried a change in code adding one line, to get it run through:

```
case 0x09: { // Long local name.
                    if (seenLongLocalName) {
                        // Prefer the long name over the short.
                        data.position(data.position() + length); // <-- new line
                        break;
                    }

```
Although I think it would be fine, to not throw the exception, but to skip the device entirely. The hardware is not setup correctly, so I guess it would be fine for the use case of this lib,

gautam-chebrolu, 2018-05-11:
Hi, I wanted to bump this back up to see if anyone was able to figure out another solution. I am trying to connect to my ESP32 dev board, and that is exactly the device that crashes the app when a scan is initiated--scanning is perfectly fine if . Running the scan by itself is fine and it produces a _ControllerStream<ScanResult> object but trying to access any part of that object or running any method throws the ArrayIndexOutOfBoundsException. I was trying to figure out if there was a way to change the BLE settings on my board but that did not work either. 
Is there a way to simply connect to board manually if I know the MAC address? I wasn't able to get it working and I read somewhere that for iOS you have to scan then connect for BLE.

MSchrenker, 2018-05-14:
@gautam-chebrolu 
1. Did you debug, where the ArrayIndexOutOfBounds happens? Is it only when running the Android App or on iPhones/iPads too? When only on Android then see my answer from April 13th.
2. We also used an ESP32 and got this Exception. Our firmware developer fixed the problem by setting the long name and the short name of the device explicitly. By default the ESP32 seem to be having two long name entries, but no short name entry. This revelead the bug in the parser of the lib.

gautam-chebrolu, 2018-05-18:
@MSchrenker 
1. Thank you for your solution! I had it running with that but I haven't been able to check how it runs on iOS yet and I wanted something that would run cross-platform. I was assuming that this would fix only Android issues.
2. I will look into editing espressif's code as well. I don't have too much experience in it, but will report back with how it works.

lemonbuzz, 2018-05-20:
Hey im having the same issue! have you found a solution?

MSchrenker, 2018-05-22:
@lemonbuzz  We modified the Arduino sample from here:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_notify/BLE_notify.ino
and replaced:
<pre>
// Start advertising
 pServer->getAdvertising()->start();
</pre>
with something like this:
<pre>// Start advertising
 //pServer->getAdvertising()->start();
 BLEAdvertising* pAdvertising = pServer->getAdvertising();
 BLEAdvertisementData data;
 data.setFlags(ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT);
 data.setName("2nd Server");
 data.setShortName("4th Server");
 pAdvertising->setAdvertisementData(data);
 pAdvertising->start();
</pre>
HTH

contango1, 2018-07-17:
@MSchrenker Thank you very much! Your solution of Apr 13 works perfect.