pauldemarco/flutter_blue #70: setNotifyValue on multiple characteristics simultaneously

psachs, 2018-04-23:
I am unable to setNotifyValue for multiple characteristics on the same device. It always crashes when I try to setNotifyValue on the second characteristic. The code to register to notifications looks like this:

```dart
void registerNotifications(Store<AppState> store, BleDevice device,
    BluetoothCharacteristic humidityCharacteristic,
    BluetoothCharacteristic temperatureCharacteristic,) async {
  await device.device.setNotifyValue(humidityCharacteristic, true);
  device.device.onValueChanged(humidityCharacteristic).listen((value) {
    device.setHumidity(LittleEndianExtractor.extractDouble(value));
    store.dispatch(new BleDeviceNewValue(device: device));
  });

  await device.device.setNotifyValue(temperatureCharacteristic, true);
  device.device.onValueChanged(temperatureCharacteristic).listen((value) {
    device.setTemperature(LittleEndianExtractor.extractDouble(value));
    store.dispatch(new BleDeviceNewValue(device: device));
  });
}
```

I get the following runtime error message

```text
D/BluetoothGatt(19458): setCharacteristicNotification() - uuid: 00001235-b38d-4985-720e-0f993a68ee41 enable: true
E/flutter (19458): [ERROR:topaz/lib/tonic/logging/dart_error.cc(16)] Unhandled exception:
E/flutter (19458): PlatformException(set_notification_error, error when writing the descriptor, null)
E/flutter (19458): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:547:7)
E/flutter (19458): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:279:18)
E/flutter (19458): <asynchronous suspension>
E/flutter (19458): #2      BluetoothDevice.setNotifyValue (file:///home/psachs/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.3.3/lib/src/bluetooth_device.dart:170:41)
E/flutter (19458): <asynchronous suspension>
E/flutter (19458): #3      registerNotifications.<anonymous closure> (package:ble_scanner/middlerware/ble_middleware.dart:166:19)
E/flutter (19458): #4      _RootZone.run (dart:async/zone.dart:1376:54)
E/flutter (19458): #5      _FutureListener.handleWhenComplete (dart:async/future_impl.dart:151:18)
E/flutter (19458): #6      Future._propagateToListeners.handleWhenCompleteCallback (dart:async/future_impl.dart:607:39)
E/flutter (19458): #7      Future._propagateToListeners (dart:async/future_impl.dart:663:37)
E/flutter (19458): #8      Future._complete (dart:async/future_impl.dart:471:7)
E/flutter (19458): #9      _SyncCompleter.complete (dart:async/future_impl.dart:51:12)
E/flutter (19458): #10     BluetoothDevice.setNotifyValue (file:///home/psachs/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.3.3/lib/src/bluetooth_device.dart)
E/flutter (19458): <asynchronous suspension>
E/flutter (19458): #11     registerNotifications (package:ble_scanner/middlerware/ble_middleware.dart:165:17)
E/flutter (19458): <asynchronous suspension>
E/flutter (19458): #12     handleDeviceConnection.<anonymous closure>.<anonymous closure> (package:ble_scanner/middlerware/ble_middleware.dart:138:15)
E/flutter (19458): #13     _RootZone.runUnary (dart:async/zone.dart:1381:54)
E/flutter (19458): #14     _FutureListener.handleValue (dart:async/future_impl.dart:129:18)
E/flutter (19458): #15     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:637:45)
E/flutter (19458): #16     Future._propagateToListeners (dart:async/future_impl.dart:666:32)
E/flutter (19458): #17     Future._complete (dart:async/future_impl.dart:471:7)
E/flutter (19458): #18     _SyncCompleter.complete (dart:async/future_impl.dart:51:12)
E/flutter (19458): #19     BluetoothDevice.discoverServices (file:///home/psachs/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.3.3/lib/src/bluetooth_device.dart)
E/flutter (19458): <asynchronous suspension>
E/flutter (19458): #20     handleDeviceConnection.<anonymous closure> (package:ble_scanner/middlerware/ble_middleware.dart:120:25)
E/flutter (19458): #21     _RootZone.runUnaryGuarded (dart:async/zone.dart:1316:10)
E/flutter (19458): #22     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:330:11)
E/flutter (19458): #23     _DelayedData.perform (dart:async/stream_impl.dart:578:14)
E/flutter (19458): #24     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:694:11)
E/flutter (19458): #25     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:654:7)
E/flutter (19458): #26     _microtaskLoop (dart:async/schedule_microtask.dart:41:21)
E/flutter (19458): #27     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50:5)
```

psachs, 2018-05-18:
It seems that the BluetoothGatt is not closed properly after setNotification but takes some time. I worked around this issue by introducing a wait time between the two registrations:
```
    registerTemperatureNotifications(store, action.device, action.characteristic);

    Future.delayed(new Duration(seconds: 1), () {
      registerHumidityNotifications(store, action.device, action.characteristic);
    });
```

I don't think 1 second delay is necessary. The register..Notifications methods  are just wrappers for the setNotifyValue

yahooooza, 2018-06-08:
Thanks @psachs, suffered from a comparable issue when trying to close a bunch of Notifications. I tried to fix the issues with the right use of asnyc and await construct, but this doesn't work for me. A delay of 400 Ms working out well for me.

@nevergoneok, I doesn't understand your explanation of this issues. Can you explain what you mean?

litong1314, 2018-07-02:
I have the same probleme.

yahooooza, 2018-09-07:
Hi, how do i get the latest commit in my flutter project without a new version number. I did flutter upgrade, that should update all packages but i am not sure if that get me the last commit.

wkornewald, 2018-09-12:
I've tested it many times and this seems to fix the issue for us. Thanks!

pauldemarco, 2018-10-09:
Seems to be fixed in version 0.4.0

closed 2018-10-09 by pauldemarco

Nailer, 2019-11-20:
I think I'm running into this problem again with flutter_blue 6.2 on my Pixel 2 running Android 10.

marileen-jansen, 2019-12-09:
> I think I'm running into this problem again with flutter_blue 6.2 on my Pixel 2 running Android 10.
> 
> EDIT: After some more testing I can confirm that I had to delay the calls between setNotifyValue an arbitrary amount until it started working.

I also have this issue on fultter_blue 0.6.2

Alex-Evo, 2020-04-28:
Same here with 0.7.1. I had to introduce a small delay (100 ms was sufficient, didn't try to go lower) to get it working on a Pixel 3a running Android 10.

carolinapdfaria, 2020-05-17:
> Same here with 0.7.1. I had to introduce a small delay (100 ms was sufficient, didn't try to go lower) to get it working on a Pixel 3a running Android 10.

Hey, how did you fix it with the delay? I still have the same error

Alex-Evo, 2020-05-27:
@carolinapdfaria sorry for the late reply, I have been super busy with an urgent project in the last 2 weeks.

Here's how I do it:
```
/// initialize by getting object representations of all the characteristics
Future<void> initialize(BluetoothService remoteBluetoothService) async {
    this.remoteBluetoothService = remoteBluetoothService;

    // get all characteristics
    final characteristics = remoteBluetoothService.characteristics;

    // map characteristic object to named characteristics
    // enabled notifying for those flagged as notifying
    await Future.forEach(characteristics, (BluetoothCharacteristic c) async {
      final uuidString = c.uuid.toString();
      if (availableServiceCharacteristics.containsKey(uuidString)) {
        final characteristicName = availableServiceCharacteristics[uuidString];
        serviceCharacteristics[characteristicName] = c;
        if (availableServiceCharacteristicNotifications[characteristicName]) {
          await Future.delayed(PulseOnDataSourceConsts.timeBetweenBleCommands,
              () async {
            await c.setNotifyValue(true);
          });
        }
      }
    });
  }
```

`availableServiceCharacteristics` is a map containing the UUIDs of the Bluetooth characteristics that I want to interact with (key is the UUID, and value is a string I use as identifier for the characteristic in other parts of my code), and `availableServiceCharacteristicNotifications` is another map that flags which of the characteristics I want to enable notifications for (key is the `characteristicName` string that was obtained as value from the other map, and value is a boolean flag that decides if notifications should be turned on).

I had to add the ` await Future.delayed(...)` to make it work under Android 10. The constant `PulseOnDataSourceConsts.timeBetweenBleCommands`is defined as `const Duration(milliseconds: 100)``

carolinapdfaria, 2020-05-29:
Thank you !! :) meanwhile we were able to solve the problem

Silhouette2022, 2021-09-20:
What do you do to stop notifications? Simply call with false? it throws an error. 
What's the correct sequence to disconnect device? Do I have to somehow stop notifications or delete the listener? Thanks

971586331, 2022-11-26:
This problem still exists on the latest version 0.8.0, and when there are more Characteristics, it is completely inoperable

psachs, 2022-11-27:
For those who care. We switched to use flutter blue plus since this library is not updated anymore: https://github.com/boskokg/flutter_blue_plus

On the new library we don't have this problem anymore