pauldemarco/flutter_blue #73: Classic Bluetooth support

gamelaster, 2018-05-02:
Hello, I have my device which using the classic Bluetooth, communicating via Serial.
It's possible to have support for this in the library?
Thanks

ddovidenko, 2018-05-12:
definitely interested in this too

chapus, 2018-05-14:
I'm interested also in classic Bluetooth communication.  There is a ScanMode option in the scan method, but none of the options can view classic Bluetooth.  What we can do to accomplish this, the android libraries are not compatible ?

jack4git, 2018-05-14:
On iOS at least, the CoreBluetooth framework used by flutter_blue is for Bluetooth Low Energy only.

GertJOdendaal, 2018-05-15:
Really need this in my current project.

martyglaubitz, 2018-05-21:
Same here i rely on scanning for non Low Energy Devices as well

KhoomeiK, 2018-06-29:
Any update on this? I'm unable to test with an HC-05 due to lack of support.

brianschardt, 2018-07-25:
I need this for my current project as well! Any updates? Anybody know how to program this, I will pay for this to happen.

edufolly, 2018-08-20:
Hi folks, I tried to find a plugin to do the interface with an HC-05 Bluetooth adapter, but I didn't find. So I started to dev one, based on flutter-blue and I would like to get your opinions and improvements. I'm waiting for your comments. https://github.com/edufolly/flutter_bluetooth_serial

ddovidenko, 2018-08-20:
Dude way to go! I'm going to check it out today. This is very important for a project I'm working on. I saw another gentleman started working on one, but it didn't seem to get any traction. I'll check this out this week and tell you what I think!

pauldemarco, 2018-10-09:
Bluetooth Classic/Serial support will be tracked in issue #118

closed 2018-10-09 by pauldemarco