pauldemarco/flutter_blue #79: Example App crashes on write to a characteristic

ssaylanc, 2018-06-07:
Example app in the library crashes when I try to write a value.

pauldemarco, 2018-08-10:
Can you please share the log of the crash?

pauldemarco, 2018-10-09:
Going to close this until further information is provided.

closed 2018-10-09 by pauldemarco

RyanYANG52, 2018-11-01:
I have the same problem, using example app, if I tap the writeCharacteristic too quickly, it'll crash. => `PlatformException(write_characteristic_error, writeCharacteristic failed, null)` in messsage_codecs.dart , decodeEnvlope method.

I also write a test app, after call discoverServices and found the right service, writeCharacteristic  immediately will cause the same error.
```
      _characteristic = _service.characteristics.first;

      _device.setNotifyValue(_characteristic, true);

      _valueChangedSubscription = _device
          .onValueChanged(_characteristic)
          .listen((d) => _handleRxData(d));

     // await Future.delayed(const Duration(milliseconds: 100));

     _device.writeCharacteristic(_characteristic, _writeCmd);
```
Uncomment the Future.delayed line, it works without error

aBuder, 2020-05-13:
Is any update here?