pauldemarco/flutter_blue #82: Installation failed on osx

ssarangi, 2018-06-19:
On installation I get the following message and the installation fails. The protoc compiler is not found...

** /Users/satyajits/dev/flutter/bluetooth_example/Pods/!ProtoCompiler/protoc         --objc_out=gen         --proto_path=../protos         ../protos/*.proto**

However, the correct path is **/Users/satyajits/dev/flutter/bluetooth_example/ios/Pods/!ProtoCompiler/protoc**

The difference is that the installation happens inside the ios folder of the flutter directory but somehow it is looking in the parent directory.

Launching lib/main.dart on iPhone X in debug mode...
Warning: CocoaPods recommended version 1.5.0 or greater not installed.
Pods handling may fail on some projects involving plugins.
To upgrade:
  brew upgrade cocoapods
  pod setup

Running pod install...
CocoaPods' output:
↳
      Preparing
    
    Analyzing dependencies
    
    Inspecting targets to integrate
      Using `ARCHS` setting to build architectures of target `Pods-Runner`: (``)
    
    Fetching external sources
    -> Fetching podspec for `Flutter` from `.symlinks/flutter/ios`
    -> Fetching podspec for `flutter_blue` from `.symlinks/plugins/flutter_blue/ios`
    
    Resolving dependencies of `Podfile`
    
    Comparing resolved specification to the sandbox manifest
      A !ProtoCompiler
      A Flutter
      A Protobuf
      A flutter_blue
    
    Downloading dependencies
    
    -> Installing !ProtoCompiler (3.5.0)
      > Copying !ProtoCompiler from `/Users/satyajits/Library/Caches/CocoaPods/Pods/Release/!ProtoCompiler/3.5.0-3e9dc` to `Pods/!ProtoCompiler`
    
    -> Installing Flutter (1.0.0)
    
    -> Installing Protobuf (3.6.0)
      > Copying Protobuf from `/Users/satyajits/Library/Caches/CocoaPods/Pods/Release/Protobuf/3.6.0-0fc0a` to `Pods/Protobuf`
    
    -> Installing flutter_blue (0.0.1)
     > Running prepare command
       $ /bin/bash -c  set -e mkdir -p gen /Users/satyajits/dev/flutter/bluetooth_example/Pods/!ProtoCompiler/protoc         --objc_out=gen         --proto_path=../protos         ../protos/*.proto
    [!] /bin/bash -c 
    set -e
    mkdir -p gen
    /Users/satyajits/dev/flutter/bluetooth_example/Pods/!ProtoCompiler/protoc         --objc_out=gen         --proto_path=../protos         ../protos/*.proto
    
    /bin/bash: line 3: /Users/satyajits/dev/flutter/bluetooth_example/Pods/!ProtoCompiler/protoc: No such file or directory
    
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/executable.rb:69:in `execute_command'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/executable.rb:27:in `block in executable'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer/pod_source_preparer.rb:66:in `block (2 levels) in run_prepare_command'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer/pod_source_preparer.rb:60:in `chdir'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer/pod_source_preparer.rb:60:in `block in run_prepare_command'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/user_interface.rb:64:in `section'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer/pod_source_preparer.rb:59:in `run_prepare_command'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer/pod_source_preparer.rb:37:in `prepare!'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer/pod_source_installer.rb:61:in `install!'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer.rb:365:in `install_source_of_pod'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer.rb:330:in `block (2 levels) in install_pod_sources'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/user_interface.rb:85:in `titled_section'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer.rb:329:in `block in install_pod_sources'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer.rb:321:in `each'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer.rb:321:in `install_pod_sources'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer.rb:164:in `block in download_dependencies'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/user_interface.rb:64:in `section'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer.rb:162:in `download_dependencies'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/installer.rb:117:in `install!'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/command/install.rb:41:in `run'
    /Library/Ruby/Gems/2.3.0/gems/claide-1.0.2/lib/claide/command.rb:334:in `run'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/lib/cocoapods/command.rb:52:in `run'
    /Library/Ruby/Gems/2.3.0/gems/cocoapods-1.4.0/bin/pod:55:in `<top (required)>'
    /usr/local/bin/pod:22:in `load'
    /usr/local/bin/pod:22:in `<main>'
Error output from CocoaPods:
↳
       /bin/bash: line 3: /Users/satyajits/dev/flutter/bluetooth_example/Pods/!ProtoCompiler/protoc: No such file or directory
    
    [!] `<PBXGroup UUID=`97C146E51CF9000F007C117D`>` attempted to initialize an object with an unknown UUID. `CF3B75C9A7D2FA2A4C99F110` for attribute: `children`. This can be the result of a merge and  the unknown UUID is being discarded.
Error running pod install
Error launching application on iPhone X.

NaelAwayes, 2018-07-03:
Experiencing the same issue.

Did Step 1 from #40 without any result. Doing step 2 enables me to build the app on Xcode, but `flutter run` still fails.

psycura, 2018-12-10:
Somebody solved this?

psycura, 2018-12-23:
@NaelAwayes How can i get the updated version of the plugin?
I still see only 0.4.1 version in the store

NaelAwayes, 2018-12-24:
Hey,

It hasn’t been merged (yet ?) so you will have to fix it locally after downloading the most recent version.

pauldemarco, 2018-12-29:
Hi everyone,

I've tested in xcode 9.4 and just updated to 10.1, and I am unable to reproduce this issue.
Please try with the latest version of FlutterBlue **0.4.2**

bradselw, 2019-01-06:
This is still an issue for me on 0.4.2.

bradselw, 2019-01-09:
Note, copying the flutter_blue repo into my project's repo and applying #89 fixed the issue for me. I would prefer not to maintain a fork of the flutter_blue repo - what is the hold up on merging #89?

bsheen, 2019-02-15:
As per @bradselw comment above, any chance we can get #89 merged in?

Andy1984, 2019-02-27:
Still exists on 0.5.0

bsheen, 2019-02-27:
@Andy1984 this issue is fixed in this [PR](https://github.com/pauldemarco/flutter_blue/pull/195)

bobdeng, 2019-03-30:
modify flutter_blue/ios/flutter_blue.podspec
change   protoc = ENV['PWD'] + 'Pods/!ProtoCompiler/protoc'
to protoc = ENV['PWD'] + '/ios/Pods/!ProtoCompiler/protoc'

works for me

Gollor, 2019-04-02:
A very simply work-around is to execute `ln -s ios/Pods Pods` in the project root. This would link Pods/ folder to the working directory so the ios/ could be omitted.

ch-bitrise, 2019-04-03:
@Gollor are you able to confirm this [PR](https://github.com/pauldemarco/flutter_blue/pull/195)

@pauldemarco has indicated he'll merge this in should he get some confirmation. The fix should resolve this for all.

Dalmangyi, 2019-08-04:
Thx @Gollor !! solved.

raspberrypisig, 2019-08-08:
I can also confirm that @Gollor fix works.

0x77dev, 2019-09-19:
> A very simply work-around is to execute `ln -s ios/Pods Pods` in the project root. This would link Pods/ folder to the working directory so the ios/ could be omitted.

Thank you very much!

pauldemarco, 2019-10-18:
This should be resolved in the latest version 0.6.3

If everyone could give it a shot and let me know, we might be able to squash this!

pauldemarco, 2020-04-02:
Protoc is no longer a dependency as the files are pre-generated.
This should no longer be an issue.

closed 2020-04-02 by pauldemarco