pauldemarco/flutter_blue #84: [Android] Crashes when Bluetooth Adapter is turned off

ened, 2018-06-28:
```
FATAL EXCEPTION: main
Process: com.company.app, PID: 11091
java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1452, result=-1, data=Intent { act=android.content.pm.action.R
EQUEST_PERMISSIONS (has extras) }} to activity {com.company.app/com.company.app.MainActivity}: java.lang.IllegalArgumentException: Unsupported 
value: java.lang.IllegalStateException: getBluetoothLeScanner() is null. Is the Adapter on?
    at android.app.ActivityThread.deliverResults(ActivityThread.java:4339)
    at android.app.ActivityThread.handleSendResult(ActivityThread.java:4382)
    at android.app.ActivityThread.-wrap19(Unknown Source:0)
    at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1654)
    at android.os.Handler.dispatchMessage(Handler.java:105)
    at android.os.Looper.loop(Looper.java:251)
    at android.app.ActivityThread.main(ActivityThread.java:6572)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
Caused by: java.lang.IllegalArgumentException: Unsupported value: java.lang.IllegalStateException: getBluetoothLeScanner() is null. Is the Adapter on?
    at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:293)
    at io.flutter.plugin.common.StandardMethodCodec.encodeErrorEnvelope(StandardMethodCodec.java:70)
    at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.error(MethodChannel.java:199)
    at com.pauldemarco.flutterblue.FlutterBluePlugin.startScan(FlutterBluePlugin.java:602)
    at com.pauldemarco.flutterblue.FlutterBluePlugin.onRequestPermissionsResult(FlutterBluePlugin.java:495)
    at io.flutter.app.FlutterPluginRegistry.onRequestPermissionsResult(FlutterPluginRegistry.java:175)
    at io.flutter.app.FlutterActivityDelegate.onRequestPermissionsResult(FlutterActivityDelegate.java:132)
    at io.flutter.app.FlutterActivity.onRequestPermissionsResult(FlutterActivity.java:134)
    at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:7388)
    at android.app.Activity.dispatchActivityResult(Activity.java:7239)
    at android.app.ActivityThread.deliverResults(ActivityThread.java:4335)
    ... 9 more
```

Plugin should query for Adapter status first and turn it on (Android will prompt the user) before scanning.

ened, 2018-06-28:
I see this was intentional in #50, but I would expect it to work the other way. ;-)

suparnavg, 2019-02-01:
@ened - did you figure out a way around this? Is there a way to turn the adapter on in case it is off?

andretietz, 2019-02-04:
This (or similar) code should be part of the plugin.

suparnavg, 2019-02-04:
@ened thanks for the tip!