pauldemarco/flutter_blue #85: how to get all scanrecord byte[] values, and onCharacteristicChanged() listener

francisraj, 2018-06-29:
hi, 
  i'm trying to use flutter blue package to develop an app (android and ios), i can not take advertisement data (rawdata scanrecord byte[] ). 
=> how to get all value of raw data?,
=> and is it possible to take raw data both android and ios?,
=> is there any method characteristic change listener like android code (onCharacteristicChanged()) ?,
to handle my read, write and notification changes.
 
could you please help me..,

ycktw, 2018-06-30:
this code works on my iphone6s and samsung a8. i am not sure what your problem is. but my opionion is to develope on android 1st and then test on iphone.

francisraj, 2018-07-05:
ya, this code works on my android mobiles

This is taken from nrf connect application:
[image: nrf scan result]
<https://user-images.githubusercontent.com/40693247/42135919-0096b996-7d70-11e8-8aea-34cc9310fd76.PNG>
in this image,when i scan ble device using nrf connect, i can get all
advertisement raw bytes data.

how to get full advertisement data bytes from flutter , i'm new for flutter,
could you please let me know on which method i can get raw data on scan
device.

pauldemarco, 2018-10-09:
Closing this, please re-open if this is still an issue.

closed 2018-10-09 by pauldemarco