pauldemarco/flutter_blue #86: get RSSI

litong1314, 2018-07-02:
is it possible to get RSSI of a CONNECTED device?

GauravPatni, 2019-01-15:
Need help on same issue .... How to get RSSI value ?

alextekartik, 2019-01-24:
Android (https://developer.android.com/reference/android/bluetooth/BluetoothGatt#readRemoteRssi()) seems to support it. and I guess iOS. I could help in the implementation (PR) if needed/wanted/accepted

LiveRock, 2019-07-09:
Yes, I need to get RSSI for connected device.

thomasgarrison, 2019-07-25:
Had anyone made progress on this? I can jump in and help if needed

icaglar, 2019-08-05:
Yes I need help on same

icaglar, 2019-08-05:
You can watch rssi values like below
FlutterBlue.instance.scanResults.listen((sr){
            sr.forEach((f){
               print("Scan Results:"+f.device.id.id.toString()+" - "+ f.rssi.toString());
            });
             
          });

icaglar, 2019-08-30:
https://github.com/icaglar/flutter_blue
I created a pull request for this repository which provides getting rssi value of the connected device for ios
After clone this repository you have to recreate protos with the following command 
in protos folder
protoc --dart_out=../lib/gen/ ./flutterblue.proto 

After that you can access rssi value of the connected device  like this

 FlutterBlue.instance.connectedDevices.then((deviceList){
                  deviceList.forEach((d) async {
                    print("RSSI from device "+d.rssi.toString());
                  });
  });

SurabhiS22, 2020-06-11:
> Had anyone made progress on this? I can jump in and help if needed

yes  pls , if possible can u pls help me in flutter bluetooth the RSSI part

SurabhiS22, 2020-06-11:
> protoc --dart_out=../lib/gen/ ./flutterblue.proto

it doesn't worked , I did tried to recreate the protos but it says "Plugin failed with status code 127."

kur0s4ki, 2020-12-16:
I have the same issue , I need to get the rssi value in Connected State. (On Android).
has anyone made any progression ?

icaglar, 2020-12-16:
> > protoc --dart_out=../lib/gen/ ./flutterblue.proto
> 
> it doesn't worked , I did tried to recreate the protos but it says "Plugin failed with status code 127."

I think the issue related with your proto plugin. remove the plugin and install again

jamal-Bahammou, 2020-12-16:
I have the same issue, I need to retrieve rssi from connected devices, need help ?

Med0808, 2020-12-16:
same thing for me ; I need to access the rssi value on android (connected)

icaglar, 2020-12-16:
I will share android side in a few days

MouradElmehammedi, 2020-12-16:
i need to get the RSSI value on android any help?

kur0s4ki, 2020-12-16:
> 
> 
> I will share android side in a few days

Thank you so much.

icglr, 2020-12-18:
I have published android support .

farhantandia, 2022-06-08:
where is the solution? I still couldn't find it :\

chhamed, 2023-05-08:
is it possible to get RSSI without connecting to the device ?