pauldemarco/flutter_blue #87: crashing while doing a scan

mistertennistik, 2018-07-04:
**I'm trying to seek for BLE device by doing a scan  using the very simple flutter demo app:** 

`import 'package:flutter/material.dart';

import 'package:flutter_blue/flutter_blue.dart';

void main(){

  runApp(new MyApp());

}
class MyApp extends StatelessWidget {

  // This widget is the root of your application.
  @override
  Widget build(BuildContext context) {
    return new MaterialApp(
      title: 'Flutter Demo',
      theme: new ThemeData(
        // This is the theme of your application.
        //
        // Try running your application with "flutter run". You'll see the
        // application has a blue toolbar. Then, without quitting the app, try
        // changing the primarySwatch below to Colors.green and then invoke
        // "hot reload" (press "r" in the console where you ran "flutter run",
        // or press Run > Flutter Hot Reload in IntelliJ). Notice that the
        // counter didn't reset back to zero; the application is not restarted.
        primarySwatch: Colors.blue,
      ),
      home: new MyHomePage(title: 'Flutter Demo Home Page'),
    );
  }
}

class MyHomePage extends StatefulWidget {
  MyHomePage({Key key, this.title}) : super(key: key);

  // This widget is the home page of your application. It is stateful, meaning
  // that it has a State object (defined below) that contains fields that affect
  // how it looks.

  // This class is the configuration for the state. It holds the values (in this
  // case the title) provided by the parent (in this case the App widget) and
  // used by the build method of the State. Fields in a Widget subclass are
  // always marked "final".

  final String title;

  @override
  _MyHomePageState createState() => new _MyHomePageState();
}

class _MyHomePageState extends State<MyHomePage> {
  int _counter = 0;

  FlutterBlue flutterBlue = FlutterBlue.instance;

  String scanRES;

  void _scan() {
    setState(() {

      var scanSubscription = flutterBlue.scan().listen((scanResult){

        scanRES = scanResult.toString();
      }

      );
      scanSubscription.cancel();
    });
  }

  @override
  Widget build(BuildContext context) {
    // This method is rerun every time setState is called, for instance as done
    // by the _incrementCounter method above.
    //
    // The Flutter framework has been optimized to make rerunning build methods
    // fast, so that you can just rebuild anything that needs updating rather
    // than having to individually change instances of widgets.
    return new Scaffold(
      appBar: new AppBar(
        // Here we take the value from the MyHomePage object that was created by
        // the App.build method, and use it to set our appbar title.
        title: new Text(widget.title),
      ),
      body: new Center(
        // Center is a layout widget. It takes a single child and positions it
        // in the middle of the parent.
        child: new Column(
          // Column is also layout widget. It takes a list of children and
          // arranges them vertically. By default, it sizes itself to fit its
          // children horizontally, and tries to be as tall as its parent.
          //
          // Invoke "debug paint" (press "p" in the console where you ran
          // "flutter run", or select "Toggle Debug Paint" from the Flutter tool
          // window in IntelliJ) to see the wireframe for each widget.
          //
          // Column has various properties to control how it sizes itself and
          // how it positions its children. Here we use mainAxisAlignment to
          // center the children vertically; the main axis here is the vertical
          // axis because Columns are vertical (the cross axis would be
          // horizontal).
          mainAxisAlignment: MainAxisAlignment.center,
          children: <Widget>[
            new Text(
              'scan results  : : ',
            ),
            new Text(
              '$scanRES',
              style: Theme.of(context).textTheme.display1,
            ),
          ],
        ),
      ),
      floatingActionButton: new FloatingActionButton(
        onPressed: _scan,
        tooltip: 'Increment',
        child: new Icon(Icons.looks),
      ), // This trailing comma makes auto-formatting nicer for build methods.
    );
  }
}
`

**but when pushing the button in order to scan in the app, it crashes.**

D/BluetoothAdapter(30983): STATE_ON
D/BluetoothLeScanner(30983): onClientRegistered() - status=0 clientIf=12
I/FlutterActivityDelegate(30983): onResume setting current activity to this
I/HwSecImmHelper(30983): mSecurityInputMethodService is null
I/HwSecImmHelper(30983): mSecurityInputMethodService is null
D/AndroidRuntime(30983): Shutting down VM
E/AndroidRuntime(30983): FATAL EXCEPTION: main
E/AndroidRuntime(30983): Process: com.aerospa.bletest2, PID: 30983
E/AndroidRuntime(30983): java.lang.ArrayIndexOutOfBoundsException: Not enough data.
E/AndroidRuntime(30983): 	at com.pauldemarco.flutterblue.AdvertisementParser.parse(AdvertisementParser.java:60)
E/AndroidRuntime(30983): 	at com.pauldemarco.flutterblue.ProtoMaker.from(ProtoMaker.java:30)
E/AndroidRuntime(30983): 	at com.pauldemarco.flutterblue.FlutterBluePlugin$2.onScanResult(FlutterBluePlugin.java:625)
E/AndroidRuntime(30983): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:355)
E/AndroidRuntime(30983): 	at android.os.Handler.handleCallback(Handler.java:743)
E/AndroidRuntime(30983): 	at android.os.Handler.dispatchMessage(Handler.java:95)
E/AndroidRuntime(30983): 	at android.os.Looper.loop(Looper.java:150)
E/AndroidRuntime(30983): 	at android.app.ActivityThread.main(ActivityThread.java:5621)
E/AndroidRuntime(30983): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(30983): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:794)
E/AndroidRuntime(30983): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:684)
I/Process (30983): Sending signal. PID: 30983 SIG: 9
Lost connection to device.

Thank you for helping

assumptionsoup, 2018-07-06:
This could be https://github.com/pauldemarco/flutter_blue/issues/66 again

rickyo-CI, 2018-07-13:
I'm having the same issue testing on a Nexus 5X running Android 8.1. Any fixes available?

assumptionsoup, 2018-07-13:
Try to see if https://github.com/pauldemarco/flutter_blue/pull/90 fixes it for you.

pauldemarco, 2018-08-10:
This looks to be caused by any devices that advertise more than 1 "Complete Local Name 0x09" packet.  PR #90 should fix this.

I will close this for now, please re-open if the problem persists.

closed 2018-08-10 by pauldemarco