pauldemarco/flutter_blue #88: Accessing Bluetooth Serial Profile

mumairofficial, 2018-07-04:
This is not the issue I need help; will this plugin help me to connected with bluetooth devices with serial porfile such as arduino etc.

brianalois, 2018-07-23:
Yes how do we scan for devices that are not low energy, and how would we access the serial port?

bigship, 2018-08-14:
For iOS platform, flutter_blue use the native CoreBluetooth framework under the hood, and CoreBluetooth only targets BlueTooth Low Energy. So I don't think it's possible.

edufolly, 2018-08-20:
Hi folks, I tried to find a plugin to do the interface with an HC-05 Bluetooth adapter, but I didn't find. So I started to dev one, based on flutter-blue and I would like to get your opinions and improvements. I'm waiting for your comments. https://github.com/edufolly/flutter_bluetooth_serial

stancecoke, 2020-10-23:
My HM10 Module offers the service 0000FFE0-0000-1000-8000-00805F9B34FB. This has read and write properties to receive and send the UART messages.