pauldemarco/flutter_blue #93: Lib example crashing with itag

apiel, 2018-08-07:
When I try your example, if my itag is on, the app is crashing: https://www.aliexpress.com/item/Original-Anti-lost-Alarm-Smart-Tag-Wireless-Bluetooth-Tracker-Child-Bag-Wallet-Key-Finder-GPS-Locator/32837341292.html

When it is off, I can see the other Bluetooth device around.

> I/flutter (11658): manufacturerData: [45, 1, 2, 0, 1, 16, 230, 243, 17, 150, 239, 144, 77, 175, 169, 252, 223, 210, 240, 84, 188, 27, 157, 233, 73, 172, 225, 148]
> I/flutter (11658): serviceData: {}
> I/flutter (11658): localName:
> I/flutter (11658): manufacturerData: [76, 0, 2, 21, 80, 118, 92, 183, 217, 234, 78, 33, 153, 164, 250, 135, 150, 19, 164, 146, 128, 123, 86, 27, 206]
> I/flutter (11658): serviceData: {}
> I/flutter (11658): localName:
> I/flutter (11658): manufacturerData: [6, 0, 1, 14, 33, 0, 32, 91, 151, 59, 198, 116, 65, 84, 45, 86, 73, 69, 45, 72, 85, 66, 48, 49]
> I/flutter (11658): serviceData: {}
> D/AndroidRuntime(11658): Shutting down VM
> E/AndroidRuntime(11658): FATAL EXCEPTION: main
> E/AndroidRuntime(11658): Process: com.pauldemarco.flutterblueexample, PID: 11658
> E/AndroidRuntime(11658): java.lang.ArrayIndexOutOfBoundsException: Not enough data.
> E/AndroidRuntime(11658): 	at com.pauldemarco.flutterblue.AdvertisementParser.parse(AdvertisementParser.java:60)
> E/AndroidRuntime(11658): 	at com.pauldemarco.flutterblue.ProtoMaker.from(ProtoMaker.java:30)
> E/AndroidRuntime(11658): 	at com.pauldemarco.flutterblue.FlutterBluePlugin$2.onScanResult(FlutterBluePlugin.java:625)
> E/AndroidRuntime(11658): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:596)
> E/AndroidRuntime(11658): 	at android.os.Handler.handleCallback(Handler.java:789)
> E/AndroidRuntime(11658): 	at android.os.Handler.dispatchMessage(Handler.java:98)
> E/AndroidRuntime(11658): 	at android.os.Looper.loop(Looper.java:164)
> E/AndroidRuntime(11658): 	at android.app.ActivityThread.main(ActivityThread.java:6938)
> E/AndroidRuntime(11658): 	at java.lang.reflect.Method.invoke(Native Method)
> E/AndroidRuntime(11658): 	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
> E/AndroidRuntime(11658): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)
> Lost connection to device.

This device are very minimalist and there might be some information missing. I don't if it's the example failing or the library. Hopefully I will find time to figure this out.

apiel, 2018-08-08:
In flutter_blue/android/src/main/java/com/pauldemarco/flutterblue/AdvertisementParser.java line 60, you throw an error without to use the error. Why not ignore the AdvertisementData, since other information are available from the device? You could as catch the error and report it in the device object? Also, this error should not make the scan failed and then the app crash...

[image]

You even have a if statement in the parent function

[image]

pauldemarco, 2018-08-10:
This looks to be caused by any devices that advertise more than 1 "Complete Local Name 0x09" packet.  PR #90 should fix this.

I will close this for now, please re-open if the problem persists.

closed 2018-08-10 by pauldemarco