pauldemarco/flutter_blue #98: Running flutter_blue on iPhone failed at pod install

bigship, 2018-08-13:
Hello everyone, I'm new to Flutter. 
I created a simple Flutter app using flutter_blue and just want to test it on real device. But every time I run the app I got the following output on debug console. Anyone know how to solve the problem ?

Launching lib/main.dart on iPhone in debug mode...
Signing iOS app for device deployment using developer identity: "iPhone Developer: Cheng Chen (XXXXXXXXXX)"
CocoaPods' output:
↳
Preparing
Analyzing dependencies
Inspecting targets to integrate
Using `ARCHS` setting to build architectures of target `Pods-Runner`: (``)
Fetching external sources
-> Fetching podspec for `Flutter` from `.symlinks/flutter/ios`
-> Fetching podspec for `flutter_blue` from `.symlinks/plugins/flutter_blue/ios`
Resolving dependencies of `Podfile`
Comparing resolved specification to the sandbox manifest
A !ProtoCompiler
A Flutter
A Protobuf
A flutter_blue
Downloading dependencies
-> Installing !ProtoCompiler (3.5.0)
> Copying !ProtoCompiler from `/Users/chenge/Library/Caches/CocoaPods/Pods/Release/!ProtoCompiler/3.5.0-3e9dc` to `Pods/!ProtoCompiler`
-> Installing Flutter (1.0.0)
-> Installing Protobuf (3.6.0)
> Copying Protobuf from `/Users/chenge/Library/Caches/CocoaPods/Pods/Release/Protobuf/3.6.0-0fc0a` to `Pods/Protobuf`
-> Installing flutter_blue (0.0.1)
> Running prepare command
$ /bin/bash -c  set -e mkdir -p gen /Users/chenge/Documents/Project_backup/flutter_course/Pods/!ProtoCompiler/protoc         --objc_out=gen         --proto_path=../protos         ../protos/*.proto
[!] /bin/bash -c
set -e
mkdir -p gen
/Users/chenge/Documents/Project_backup/flutter_course/Pods/!ProtoCompiler/protoc         --objc_out=gen         --proto_path=../protos         ../protos/*.proto
/bin/bash: line 3: /Users/chenge/Documents/Project_backup/flutter_course/Pods/!ProtoCompiler/protoc: No such file or directory
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/executable.rb:69:in `execute_command'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/executable.rb:27:in `block in executable'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer/pod_source_preparer.rb:66:in `block (2 levels) in run_prepare_command'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer/pod_source_preparer.rb:60:in `chdir'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer/pod_source_preparer.rb:60:in `block in run_prepare_command'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/user_interface.rb:64:in `section'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer/pod_source_preparer.rb:59:in `run_prepare_command'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer/pod_source_preparer.rb:37:in `prepare!'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer/pod_source_installer.rb:61:in `install!'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer.rb:374:in `install_source_of_pod'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer.rb:339:in `block (2 levels) in install_pod_sources'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/user_interface.rb:85:in `titled_section'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer.rb:338:in `block in install_pod_sources'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer.rb:321:in `each'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer.rb:321:in `install_pod_sources'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer.rb:164:in `block in download_dependencies'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/user_interface.rb:64:in `section'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer.rb:162:in `download_dependencies'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/installer.rb:117:in `install!'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/command/install.rb:41:in `run'
/usr/local/lib/ruby/gems/2.5.0/gems/claide-1.0.2/lib/claide/command.rb:334:in `run'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/lib/cocoapods/command.rb:52:in `run'
/usr/local/lib/ruby/gems/2.5.0/gems/cocoapods-1.5.0/bin/pod:55:in `<top (required)>'
/usr/local/bin/pod:23:in `load'
/usr/local/bin/pod:23:in `<main>'
Error output from CocoaPods:
↳
WARNING: CocoaPods requires your terminal to be using UTF-8 encoding.
Consider adding the following to ~/.profile:
export LANG=en_US.UTF-8

/bin/bash: line 3: /Users/chenge/Documents/Project_backup/flutter_course/Pods/!ProtoCompiler/protoc: No such file or directory
[!] `<PBXGroup UUID=`97C146E51CF9000F007C117D`>` attempted to initialize an object with an unknown UUID. `CF3B75C9A7D2FA2A4C99F110` for attribute: `children`. This can be the result of a merge and  the unknown UUID is being discarded.
[!] Automatically assigning platform `ios` with version `8.0` on target `Runner` because no platform was specified. Please specify a platform for this target in your Podfile. See `https://guides.cocoapods.org/syntax/podfile.html#platform`.
Error running pod install
Exited (sigterm)

ycktw, 2018-08-14:
> /bin/bash: line 3: /Users/chenge/Documents/Project_backup/flutter_course/Pods/!ProtoCompiler/protoc: No such file or directory
> [!] <PBXGroup UUID=97C146E51CF9000F007C117D> attempted to initialize an [...]

Have you already signed up a apple developer id ??

bigship, 2018-08-14:
Solved. I need to setup a valid Signing Certificate and provisioning profile in Xcode before running the app.

closed 2018-08-14 by bigship

dannyalbuquerque, 2019-07-01:
How did you fix this ?

> /bin/bash: line 3: /Users/chenge/Documents/Project_backup/flutter_course/Pods/!ProtoCompiler/protoc: No such file or directory