pauldemarco/flutter_blue #99: pub get failed  - Is there a workaround to install the latest version somehow?

phev8, 2018-08-15:
Hello,

I included the package into a project with freshly installed flutter (Flutter Release Preview 1 - 0.5.1), and I get the following error message:

     Because every version of flutter_test from sdk depends on convert 2.0.1 and flutter_blue 0.4.0 depends on convert ^2.0.2, flutter_test from sdk is incompatible with flutter_blue 0.4.0.

pubspec.yaml is generated by 'flutter create myapp' and I've just added flutter_blue: ^0.4.0 into the dependencies

Is there any solution/workaround for this problem.

nhardt, 2018-08-16:
A general flutter solution is to use dependency overrides.

https://www.dartlang.org/tools/pub/dependencies#dependency-overrides

in pubspec.yaml:
```
dependencies:
  flutter:
    sdk: flutter
  cupertino_icons: ^0.1.2
  flutter_blue: ^0.4.0

dependency_overrides:
  convert: ^2.0.2
```

ened, 2018-08-16:
@nhardt nice !

@phev8 alternatively, you can switch to the flutter `dev` channel, which solves the "convert" dependency issue and includes a few bug fixes.

phev8, 2018-08-16:
Thank you very much for the suggestions. I used the dependency-override for now and it helped so far.

closed 2018-08-16 by phev8