pauldemarco/flutter_blue #102: 2 Bytes missing in advertisementData.manufacturerData

yahooooza, 2018-08-28:
I tried to update flutter_blue in my Project from  0.3.3 to 0.4.0 and managed that the manufacturerData are now a Map. Now i am facing the issue that the first 2 Bytes of the Advertismentdata.manufacturerdata are missing. The example of flutter blue have the same issue. nRFConnect show me the correct Advertismentdata.manufacturerdata.

yahooooza, 2018-09-07:
Did anyone experienced the same issue? I already took a second Smartphone, but it shows the same behaviour. The first 2 Byte of the manufacturerData are missing in 0.4.0.

dokinkon, 2018-09-20:
I have same problem. Because manufacturerData is a Map in flutter. I guess first two byte was parsed as 2bytes integer and placed into Map key. and Map value field lost 1 byte.

Andy1984, 2019-02-20:
I do not understand why manufacturerData is a map. In my case, the data in xCode is <f085c1af 056d>, which is written in hexadecimal. The same data in flutter is {34288: [193, 175, 5, 109]}. The first two bytes f085, in little-endian, is actually 85f0, which is 34288 in decimal. c1 is 193, af is 175, 05 is 5, 6d is 109 in decimal. In all, the first two bytes are not missing. You should read it in little-endian and transform it to decimal. Good luck.

robomike, 2020-02-25:
Hi All, maybe I am just confused.  Here are screen shots of my device in nordic RF and flutter blue , how is the raw data presented in flutter blue ? 
[image]
[image]
[image]
[image]

dokinkon, 2020-07-12:
> The way Android and iOS handle manufacturer advertisement data is a bit different:
> 
> * Android allows access to the raw advertisement packets, from which you can extract the raw bytes.
> * iOS maps the first 2 bytes as the Company Specifier, and the rest as that company's data.
> 
> Typically the plugin would support both versions, but seeing as though the [Bluetooth Specification](https://www.bluetooth.com/specifications/bluetooth-core-specification) requires the first 2 bytes to be company specifier, I decided on that for the plugin, see here:
> 
> [image]
> 
> If it's not possible for you to change the device so that it's **compliant** with the Bluetooth Specification, please describe your use case and we can work on a solution.

You are totally right. My customer doesn't obey the BLE specification and cause this problem.

avdisx, 2021-09-07:
> The way Android and iOS handle manufacturer advertisement data is a bit different:
>
> * Android allows access to the raw advertisement packets, from which you can extract the raw bytes.
> * iOS maps the [...]

My device is sending private data in these undisplayed bytes and I cannot see this data. How can we come up with a solution? good work.