pauldemarco/flutter_blue #104: flutter blue  intl_translation failed

fjbart70, 2018-08-31:
Trying to run you example and error below.  I realize it is a setup error, not anything in your example, but I can't figure out how to load the correct sdk?  Any help appreciated.

C:\src\flutter\flutter\bin\flutter.bat --no-color packages get
Running "flutter packages get" in example...
The current Dart SDK version is 2.1.0-dev.0.0.flutter-be6309690f.

Because flutter_blue_example depends on intl_translation >=0.14.0+1 <0.17.0 which requires SDK version >=1.12.0 <2.0.0, version solving failed.
pub get failed (1)
Process finished with exit code 1

fjbart70, 2018-09-01:
Not sure how to load a lower version of DART SDK in Android Studio?
Path for DART is currently C:\src\flutter\flutter\bin\cache\dart-sdk

Any help in installing a version <2.0.0 would be appreciated.

pauldemarco, 2018-10-09:
The intl libraries are not required, and have been removed in 2b6c307793d80e1f551cd07e1725675c937ec571

closed 2018-10-09 by pauldemarco