pauldemarco/flutter_blue #105: Guid format error: "The format is invalid"

ssaylanc, 2018-09-01:
For scanning devices with a UUID like "FFF0", I receive "The format is invalid" error.

Fleximex, 2018-09-04:
I ran into a similar problem. It seems FlutterBlue uses full UUID's.
So for example I used UUID "FFE1" in a totally different Bluetooth C# framework. But FlutterBlue requires "0000ffe1-0000-1000-8000-00805f9b34fb" in my case.

A good practice in understanding UUID's, services, descriptors, etc is downloading the app 'BLE Scanner' for either Android or iOS. The app will give you all the information of the Bluetooth device in question.
https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner
https://itunes.apple.com/us/app/id1221763603

pauldemarco, 2018-10-09:
Yes, Fleximex is right, you must use full UUIDs.

closed 2018-10-09 by pauldemarco

csga5000, 2018-11-12:
This is kind of annoying, any chance we could add support?