pauldemarco/flutter_blue #106: Example App Android build connection fails, iOS succeed

ssaylanc, 2018-09-01:
Android build of example app successfully scans and finds the device, but unable to connect.
iOS build successfully connects. 
I tried LightBlue explorer on my android device. This app successfully connects the ble device.

rchampeaux, 2018-09-02:
I'm having the same problem with Android. Haven't tried iOS.
I get an advertisement and try to connect to the device, and nothing happens. All I see in the log is:
    D/BluetoothManager(26174): getConnectedDevices
    D/BluetoothGatt(26174): connect() - device: 49:AE:36:50:B9:F2, auto: true
    D/BluetoothGatt(26174): registerApp()
    D/BluetoothGatt(26174): registerApp() - UUID=65529efe-591d-4256-8f5d-42f65a53f2da
    D/BluetoothGatt(26174): onClientRegistered() - status=0 clientIf=14

champeauxr, 2018-10-25:
That's not the way Bluetooth LE is supposed to work. You don't have to pair with Bluetooth LE devices before connecting to them. What you are describing is for legacy bluetooth, such as headphones.

I ended up writing my own Bluetooth plugin instead of using FlutterBlue and was able to connect to my Bluetooth LE device just fine without pairing.

mjl0602, 2018-10-25:
Thank you very much. It seems that I didn't found correct way, however it's really works...
Could you share you Bluetooth plugin Code with me? Or you can tell me what's the problem with FlutterBlue in Android. Thank you.
非常感谢你。应该是我没有找到比较好的方法，虽然确实能用。
你可以与我分享这部分代码吗？或者向我指出FlutterBlue的问题所在，谢谢。

champeauxr, 2018-10-25:
It might have worked, but you aren't supposed to need to do that. And as the original post says, he was able to connect to the device using the LightBlue Explorer app. Likewise, with my device, I was able to connect to it with the standard Kotlin based Android app we have for it.

I don't know what's wrong with FlutterBlue and unfortunately I can't provide you with my code, as it is part of a product. I spent some time trying to figure out why FlutterBlue didn't work, but I didn't see anything that looked wrong.

gabejudkins, 2018-12-17:
I am experiencing this issue as well. I can connect with iOS devices just fine, but on Android I can only connect to a peripheral once; subsequent connections to the same peripheral fail:

```
D/BluetoothAdapter(15921): isLeEnabled(): ON
D/BluetoothLeScanner(15921): onScannerRegistered() - status=0 scannerId=8 mScannerId=0
D/BluetoothAdapter(15921): isLeEnabled(): ON
D/BluetoothManager(15921): getConnectionState()
D/BluetoothManager(15921): getConnectedDevices
D/BluetoothManager(15921): getConnectedDevices
D/BluetoothGatt(15921): connect() - device: D0:4A:6E:CC:0F:F0, auto: true
D/BluetoothGatt(15921): registerApp()
D/BluetoothGatt(15921): registerApp() - UUID=fab11535-881d-4199-bf73-5eb82fd5053c
D/BluetoothGatt(15921): onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt(15921): close()
D/BluetoothGatt(15921): unregisterApp() - mClientIf=8
```

I am using the provided sample app with some very small modifications for testing. This is what I am calling to connect to a device:

```

  _connect(BluetoothDevice d) {
    _stopScan();

    device = d;
    // Connect to device
    deviceConnection = _flutterBlue
        .connect(device, timeout: const Duration(seconds: 4))
        .listen(
        (data) => print('onData: ' + data.toString()),
      onError: (e) => print('onError: ' + e.toString()),
      onDone: _disconnect,
    );

    // Update the connection state immediately
    device.state.then((s) {
      setState(() {
        deviceState = s;
      });
    });

    // Subscribe to connection changes
    deviceStateSubscription = device.onStateChanged().listen((s) {
      print('deviceStateSubscription = device.onStateChanged(): ' + s.toString());
      setState(() {
        deviceState = s;
      });
      if (s == BluetoothDeviceState.connected) {
        device.discoverServices().then((s) {
          setState(() {
            services = s;
          });
        });
      }
    });
  }

```

And again, on iOS it works 100% of the time. On Android it will connect once, and subsequent attempts fail.

A successful connection's output:

```
D/BluetoothAdapter(15921): isLeEnabled(): ON
D/BluetoothLeScanner(15921): onScannerRegistered() - status=0 scannerId=8 mScannerId=0
D/BluetoothAdapter(15921): isLeEnabled(): ON
D/BluetoothManager(15921): getConnectionState()
D/BluetoothManager(15921): getConnectedDevices
D/BluetoothManager(15921): getConnectedDevices
D/BluetoothGatt(15921): connect() - device: D0:4A:6E:CC:0F:F0, auto: true
D/BluetoothGatt(15921): registerApp()
D/BluetoothGatt(15921): registerApp() - UUID=dda6106a-a422-453a-a395-1211d0e62b68
D/BluetoothGatt(15921): onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt(15921): onClientConnectionState() - status=0 clientIf=8 device=D0:4A:6E:CC:0F:F0
D/FlutterBluePlugin(15921): onConnectionStateChange: 
I/flutter (15921): deviceStateSubscription = device.onStateChanged(): BluetoothDeviceState.connected
I/flutter (15921): deviceStateSubscription = device.onStateChanged(): BluetoothDeviceState.connected
D/BluetoothGatt(15921): discoverServices() - device: D0:4A:6E:CC:0F:F0
I/flutter (15921): connected!
I/flutter (15921): onData: BluetoothDeviceState.connected
D/BluetoothGatt(15921): onConnectionUpdated() - Device=D0:4A:6E:CC:0F:F0 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(15921): onSearchComplete() = Device=D0:4A:6E:CC:0F:F0 Status=0
D/FlutterBluePlugin(15921): onServicesDiscovered: 6 sink:io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler$EventSinkImplementation@a0d56ea
D/BluetoothGatt(15921): onConnectionUpdated() - Device=D0:4A:6E:CC:0F:F0 interval=36 latency=0 timeout=500 status=0
```

mjl0602, 2018-12-18:
If you can’t use flutterblue connect device with Android Phone. You can try low version Api of Android. These apis are deprecated but worked very well in some Android Phone. These
Phones are bad support with new API: you can call the function successful, but scan and connect will be very slow and easy connect fail. To solve this problem, change file:
 android/src/main/java/com/pauldemarco/flutterblue/

see new file on my fork: https://github.com/mjl0602/flutter_blue.git

This change didn't solve this problem completely. To solve this problem, must add new args to control the api version. but not use Build.VERSION.SDK_INT. It's works bad.

简单的说，就是很多手机系统到了新版本，但是对新版本的硬件API支持的很差，强制换成老版本的用法就好了。希望作者加一个字段来控制具体用什么版本的API来搜索。

* Change connect way.
```java
// old code,hardly connect device on Red Mi Note 4 
// BluetoothGatt gattServer = device.connectGatt(registrar.activity(), options.getAndroidAutoConnect(), mGattCallback);

// improve MI phone connect speed. If didn't call connect,Mi phone can't connect success
BluetoothGatt gattServer = device.connectGatt(registrar.activity(), false, mGattCallback);
gattServer.connect();
```
* Use old version api
```java
// use old version api
private void startScan(MethodCall call, Result result) {
    byte[] data = call.arguments();
    Protos.ScanSettings settings;
    try {
        settings = Protos.ScanSettings.newBuilder().mergeFrom(data).build();
        // if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP) {
        //     startScan21(settings);
        // } else {
            startScan18(settings);
        // }
        result.success(null);
    } catch (Exception e) {
        result.error("startScan", e.getMessage(), e);
    }
}

private void stopScan() {
    // if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP) {
    //     stopScan21();
    // } else {
        stopScan18();
    // }
}
```
* Remove uuids arg. 
```java
// boolean success = mBluetoothAdapter.startLeScan(uuids, getScanCallback18());
boolean success = mBluetoothAdapter.startLeScan(getScanCallback18());
```

thachnb85, 2019-01-09:
Got the same issue:
```
D/BluetoothManager( 1459): getConnectionState()
D/BluetoothManager( 1459): getConnectedDevices
D/BluetoothManager( 1459): getConnectedDevices
D/BluetoothGatt( 1459): connect() - device: 56:F7:D5:38:9D:XX, auto: true
D/BluetoothAdapter( 1459): isSecureModeEnabled
D/BluetoothGatt( 1459): registerApp()
D/BluetoothGatt( 1459): registerApp() - UUID=647a493b-c127-450d-a290-86bec4926ef7
D/BluetoothGatt( 1459): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt( 1459): close()
D/BluetoothGatt( 1459): unregisterApp() - mClientIf=6
```

stuartsoft, 2019-02-27:
I'm seeing the same issue over here. Running Android 9 Pie on a Google Pixel 3 XL

stuartsoft, 2019-02-27:
https://github.com/pauldemarco/flutter_blue/issues/171

stuartsoft, 2019-02-27:
OK OK OK OK. My dudes. Just set autoconnect to false. Everything works when you set autoconnect to `false`.

thachnb85, 2019-02-28:
@stuartsoft  Changing autoconnect to false doesn't help. still quickly disconnect after few seconds.

thevanix, 2019-03-04:
> OK OK OK OK. My dudes. Just set autoconnect to false. Everything works when you set autoconnect to `false`.

Thank you

bretmh, 2019-04-03:
I am still receiving this issue with autoconnect set to false. Using the LightBlue app I am getting a hang on connecting.

ferryirawan, 2019-05-24:
> OK OK OK OK. My dudes. Just set autoconnect to false. Everything works when you set autoconnect to `false`.

where should I write auto connect false?

Ozzy340, 2019-06-17:
Disabling autoConnect fixed this issue for me too.

I found another pattern as well if it helps anyone diagnose the root cause in the future. It worked just fine with autoconnect on my flagship device, however when I ran the flutter app on a cheap device (£80 rather than £500) it would only connect 1/10 (if not less). 

Must be something to do with the ability of some devices to hold the autoConnect feature (might be something to do with whitelisting in the BLE stack, I get the impression the autoConnect feature may be there for speedy connections when a device returns back into range in advertising mode).

jtrovato, 2020-01-17:
I was having the same issue as @Ozzy340. With auto connect off, my pixel 2 running Android 9 works perfectly but my Samsung Galaxy S5 barely connects. iOS pretty much works all the time. I found that my device connected the first time pretty consistently but would not connect after that. I found that explicitly calling disconnect() on the device after receiving the disconnected status from the device. maybe this is enforcing a direct connect each time?