pauldemarco/flutter_blue #108: iOS pod install fails

magnatronus, 2018-09-05:
I have upgrade my Flutter install to v0.7.3 beta  and I now get an error when trying to install onto iOS, this was not an issue before, it appears to be something to do with cocoapods.  With a vanilla Flutter app just adding  the flutter blue dependency to the pubspec file causes the issue.

,,,
-> Installing flutter_blue (0.0.1)
> Running prepare command
$ /bin/bash -c  set -e mkdir -p gen /Users/accountone/Development/Flutter/hw/Pods/!ProtoCompiler/protoc         --objc_out=gen         --proto_path=../protos         ../protos/*.proto
[!] /bin/bash -c
set -e
mkdir -p gen
/Users/accountone/Development/Flutter/hw/Pods/!ProtoCompiler/protoc         --objc_out=gen         --proto_path=../protos         ../protos/*.proto
/bin/bash: line 3: /Users/accountone/Development/Flutter/hw/Pods/!ProtoCompiler/protoc: No such file or directory
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/executable.rb:69:in `execute_command'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/executable.rb:27:in `block in executable'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer/pod_source_preparer.rb:66:in `block (2 levels) in run_prepare_command'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer/pod_source_preparer.rb:60:in `chdir'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer/pod_source_preparer.rb:60:in `block in run_prepare_command'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/user_interface.rb:64:in `section'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer/pod_source_preparer.rb:59:in `run_prepare_command'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer/pod_source_preparer.rb:37:in `prepare!'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer/pod_source_installer.rb:61:in `install!'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer.rb:374:in `install_source_of_pod'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer.rb:339:in `block (2 levels) in install_pod_sources'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/user_interface.rb:85:in `titled_section'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer.rb:338:in `block in install_pod_sources'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer.rb:321:in `each'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer.rb:321:in `install_pod_sources'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer.rb:164:in `block in download_dependencies'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/user_interface.rb:64:in `section'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer.rb:162:in `download_dependencies'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/installer.rb:117:in `install!'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/command/install.rb:41:in `run'
/Library/Ruby/Gems/2.3.0/gems/claide-1.0.2/lib/claide/command.rb:334:in `run'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/lib/cocoapods/command.rb:52:in `run'
/Library/Ruby/Gems/2.3.0/gems/cocoapods-1.5.3/bin/pod:55:in `<top (required)>'
/usr/local/bin/pod:22:in `load'
/usr/local/bin/pod:22:in `<main>'
Error output from CocoaPods:
↳
/bin/bash: line 3: /Users/accountone/Development/Flutter/hw/Pods/!ProtoCompiler/protoc: No such file or directory
[!] Automatically assigning platform `ios` with version `8.0` on target `Runner` because no platform was specified. Please specify a platform for this target in your Podfile. See `https://guides.cocoapods.org/syntax/podfile.html#platform`.
Error running pod install
,,,

This line appers to be looking in the root dir of the project, but the Pods dir is in the iOS directory

magnatronus, 2018-09-07:
I seem to have solved my issue by editing the cached flutter_blue.podspec  and modifying the line 

    protoc = ENV['PWD'] + '/Pods/!ProtoCompiler/protoc'

to 

    protoc = ENV['PWD'] + '/ios/Pods/!ProtoCompiler/protoc'

Now it compiles and run as before.

pauldemarco, 2018-12-29:
Which version of xcode were you seeing this error on?

malleusinferni, 2019-01-21:
@pauldemarco I saw this under Xcode 10.1 (10B61) running on 10.13. I'm planning to upgrade that device to 10.14 soon. At that time, I'll try again and see if anything has mysteriously changed.

In the meantime, some casual googling suggests that relying on `ENV['PWD']` in a podspec file is extremely fragile, and that a more robust approach would use the `PODS_ROOT` variable set by CocoaPods:

```ruby
  protoc = "#{PODS_ROOT}/!ProtoCompiler/protoc"
```

I may have gotten this wrong. I'll try and find some more information, and test this the next time I have a chance.

catacs, 2019-07-17:
@pauldemarco I saw changes in the last version to the file `flutter_blue.podspec` (https://github.com/pauldemarco/flutter_blue/commits/master/ios/flutter_blue.podspec). This change have broken the build from my root project directory in ios. 

When i use `flutter run` I get:

```
   -> Installing flutter_blue (0.0.1)
     > Running prepare command
       $ /bin/bash -c  set -e mkdir -p gen /flutter_project/Pods/!ProtoCompiler/protoc         --objc_out=gen         --proto_path=../protos         ../protos/*.proto
    [!] /bin/bash -c
    set -e
    mkdir -p gen
    /flutter_project/Pods/!ProtoCompiler/protoc         --objc_out=gen         --proto_path=../protos         ../protos/*.proto

    /bin/bash: line 3: /flutter_project/Pods/!ProtoCompiler/protoc: No such file or directory
```

It seems that it's searching for `Pods` directory in the project root directory instead of  `/flutter_project/ios/Pods`.

On the other hand if a run from command line  `pod install` into `/flutter_project/ios` and then run build from project into xcode everything seems to works.

malleusinferni, 2019-07-21:
Since it may not be clear from previous discussion, this issue is the result of several underlying causes:

1. The build process for flutter_blue's iOS support is sensitive to the working directory in which `pod install` is executed (because it needs to build `!ProtoCompiler` and then shell out to it).
2. Different versions of Flutter have historically executed that build step in different working directories.
3. Projects created with old versions of Flutter contain support files that codify problem 2, even after users upgrade Flutter to the latest version.

Right now I can create a new project with a recent Flutter installation and install flutter_blue without problems. My environment is as follows:

- Flutter beta channel 1.7.8+hotfix.3
- Mac OS 10.4.5
- Xcode 10.2.1

(I doubt that the target iOS version particularly matters, since the build process runs under Mac OS, not iOS. I can build and deploy to multiple iOS versions.)

As I described [here](https://github.com/pauldemarco/flutter_blue/pull/195#issuecomment-513569146), all attempts so far to fix problem 1 have been non-starters, because they caused more problems than they solved. It seems to me that the easiest way forward for flutter_blue is to drop iOS support for both old Flutter versions and any projects *created* with old Flutter versions. As such, people still experiencing this problem, such as @catacs, should start from scratch with new support files by creating a new project with `flutter create`.

catacs, 2019-07-22:
@malleusinferni I have just tested what you are saying about create a new project with `flutter create`. I am experiencing the same problem.

My environment is:
Flutter 1.7.8+hotfix.3 • channel stable 
Mac OS 10.14.4
Xcode 10.2.1
pod --version: 1.7.5

Modifying manually the `protoc` path in the `../flutter_blue-0.6.1+1/ios/flutter_blue.podspec` file I get it working with the initial variable used to set it `protoc =  '#{PODS_ROOT}/Pods/!ProtoCompiler/protoc'`

vincentsong, 2019-10-17:
@pauldemarco  could you please explain why you revert the change in commit 38ae679f ?

I got the same problem which @catacs did have

pauldemarco, 2019-10-18:
This should be resolved in the latest version 0.6.3

If everyone could give it a shot and let me know, we might be able to squash this!

pauldemarco, 2020-04-01:
Protoc has been removed from the project, as the files are compiled before hand.
This should no longer be an issue.

closed 2020-04-01 by pauldemarco