pauldemarco/flutter_blue #112: Should this work to connect two Android phones?

Simpler1, 2018-09-09:
I installed the example app on a Samsung Galaxy S7 and it finds bluetooth devices like speakers, but it won't find other Android phones that have bluetooth enabled and are scanning.

Has anyone else been successful with this?

Thanks

nhardt, 2018-09-09:
This was the very first thing I expected to do as well. see: https://github.com/pauldemarco/flutter_blue/issues/47 

I was ultimately able to get 2 android phones to see each other. With the pull request in the commit, but it looked like you would still need to implement the server side of the implementation in the native code to be able to transfer data.

pauldemarco, 2018-10-09:
This library supports Bluetooth Low Energy at the moment. 

If you are working with Bluetooth Classic devices, please subscribe to issue #118

closed 2018-10-09 by pauldemarco