pauldemarco/flutter_blue #113: Support for background scanning

wkornewald, 2018-09-12:
It would be awesome if flutter_blue supported scanning in background, so when a device with certain properties appears the app can be waked up and do some processing in the background. This could be based on the mechanism described here:

https://github.com/flutter/flutter/issues/3671#issuecomment-417064962

The link to the iOS sample is broken in that comment and should be:
https://github.com/bkonyi/Flutter-Location-Plugin-Background

On Android 8+ background scanning could be based on ScanFilter with PendingIntent. On older Android versions this could use a background task with ScanFilter + callback. A foreground service shouldn't be necessary (though it might be useful in rare cases, e.g. when working with beacons). Some of this is also described in the AltBeacon library: http://www.davidgyoungtech.com/2017/08/07/beacon-detection-with-android-8

dude8604, 2018-09-23:
Also being able to be notified when there incoming data from a specific device or from any device.

dude8604, 2018-09-24:
I don't know if it's against the rules to post this. If it is, please delete this comment or tell me to and I will.

I put a $10 [bounty on this issue](https://www.bountysource.com/issues/63332925-support-for-background-scanning). It's probably not enough, so if anyone else wants to add to the bounty please do.

ram231, 2018-09-27:
There was a plugin for beacons. though It doesn't support background scanning.

fvisticot, 2018-12-25:
I would be very interested to have background support for this plugin. Is it in the roadmap (Android only) ?

jifalops, 2019-01-27:
I have a use case for this. I have a customer that wants to detect nearby devices with the app installed. Periodically scanning for BLE devices in the background is necessary. 

Can anyone comment if https://pub.dartlang.org/packages/beacons is a suitable workaround to enable background scanning for nearby BLE devices?

michaellee8, 2019-03-05:
@jifalops It says monitoring (include background), maybe worth a try?

JohnWeez, 2019-06-30:
Can anyone confirm if flutter is a good solution for background beacon scanning? We are running into a bunch of issues on the newest IOS.

alod2019, 2019-07-11:
There is the fix that should enable background scanning but it's not yet merged, see
https://github.com/pauldemarco/flutter_blue/issues/187
https://github.com/pauldemarco/flutter_blue/pull/191

toniree, 2020-06-02:
Is this supported as well? "Have a BLE beacon send signals 24/7, and when a phone comes within range, the Beacon polls the phone for some data, while the phone is locked.

gib40f, 2020-11-06:
This is a feature that would be very useful to a project we are working on. Would be really good to have the app woken up by a device. I have built the app to do a scan when being brought to the foreground looking for a specific device that has been paired by the user. We are getting some uphill to make this process  more seamless.

SantiagoMinian, 2021-04-12:
On our case, we only needed to add UIBackgroundModes as stated [here](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW7) in order to continue listening to BLE device updates while the phone was locked or the user had switched to a different app.