pauldemarco/flutter_blue #114: Manufacturer Advertisement Data gets overwritten

wabash9000, 2018-09-12:
The AdvertisementData parse() function is throwing out half of my manufacturer advertisement data.  My data coming from my device has 2 sections of 0xFF, one 14 bytes long, the other 20 bytes long.  The 14 byte section has the data I care about for now but is getting overwritten by the second section.

Raw Advertisement data:
2 1 6 6 9 67 76 66 76 69 15 255 0 0 77 88 48 51 0 0 0 0 10 0 0 0 21 255 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

I added a boolean flag to the function in my code to only take the first group of data, but it would probably be ideal if it joined the two sections into one larger array.

pauldemarco, 2018-10-09:
This should be the same as #102 

Please see that issue for a full explanation.

wabash9000, 2018-10-12:
That is not the same issue.  I am missing quite a bit more data than the first 2 bytes.  I updated to the latest version of Flutter and Flutter_Blue forgetting that I had made changes in the library and now it's broken again and I can't find where I made the change previously.

When I print the manufacture data like this:
`AdvertisementData advData = scanResult.advertisementData;
for (int key in advData.manufacturerData.keys) {
    List<int> manData = advData.manufacturerData[key];
    print("manufacturerData for $key: $manData");
}`
I am only getting the result:
`manufacturerData for 0: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]`
I am expecting the full array that is in my initial post.

I remember that the data was multiple chunks and it was overwriting the previous chunk every time a new chunk came in rather than appending to the end of the array.

wabash9000, 2018-10-12:
I am testing on Android 7.1.1 on a Lenovo Tab 4 8".   The data in the first post is what is printed in xcode when print the manufacture data in swift.  

In the previous version of Flutter_blue I modified the library to only accept the first chuck of the advertisement data.  I tried to do the same modification to AdvertisementParser.java but I'm not having luck getting it to work this time.

After the modification I was previously able to pull out the data I needed with: 
`String fwVer = decoder.convert( advData.manufacturerData, 2, 5);`
`for (int i = 6; i < 10; i++) {`
`uid = uid << 8;`
`uid += advData.manufacturerData[i];`
`}`

Thanks for your help.

wabash9000, 2019-01-03:
Figured out what was happening.  We had the Scan Response Packet enabled.  We had intended to store different information in that packet, but never implemented that feature.  When the scan response packet was enabled, the manufacture specific data in it would overwrite the manufacture specific data from the advertisement packet.  On iOS it would join the two byte arrays into one, but the android library was over-writing it.  

Using the older target API I had been able to modify "AdvertisementData parse(byte[] rawData) " to join the arrays, but with target API 21 it was removed from flutter blue to the android library, so I can't work around it.  Disabled the scan response packet and now I'm able to see the data I care about.

closed 2019-01-03 by wabash9000

avdisx, 2021-09-02:
> scan response packet

hi wabash. how i can for see 2 bytes? thank you.

LeoAiolia, 2022-03-28:
> Figured out what was happening. We had the Scan Response Packet enabled. We had intended to store different information in that packet, but never implemented that feature. When the scan response [...]

How to resolve this issue?

LeoAiolia, 2022-03-28:
> How to resolve this issue?

@wabash9000