pauldemarco/flutter_blue #118: Bluetooth Classic support

pauldemarco, 2018-10-09:
Some users would like this library to support Bluetooth Classic devices, such as serial.

This will be the main issue to track support for Bluetooth Classic.

risk1996, 2018-11-20:
Any progress on Bluetooth Classic serial support? I am going to use this library to connect to an ESP 32 device.

pauldemarco, 2019-01-20:
> I'd also be willing to chip in. I really need regular BT support. Not just
> BLE

Let's get a discussion going about use cases:

What type of device/profile would you like to communicate with?

stuartsoft, 2019-01-21:
I have a use case for a personal project I'm working on. I'd like to connect to a RPI3 and transfer data. I'd prefer to use a classic bluetooth connection, as BLE requires the user to have newer hardware.

Simpler1, 2019-01-21:
I'd like to create a mobile app where multiple modern smartphones can communicate through classic Bluetooth.

stuartsoft, 2019-01-26:
Actually on second thought, I think I've got my RPI3 working with BLE. I'm good with just using Flutter_Blue now :)

ened, 2019-02-02:
@kerberjg iOS devices do connect using Bluetooth Classic, although it must be a MFi certified device in order to connect. https://stackoverflow.com/a/18214742/375209. The changes would mostly require the introduction of a EAAccessory handler in the plugin.

In general, a few things would be different with BTc
1. Scanning for devices on Android is a little different. You submit a scan request and listen for intents (`android.bluetooth.adapter.action.DISCOVERY_FINISHED`). On iOS, you let the system scan and prompt the user for the device they want to connect to https://developer.apple.com/documentation/externalaccessory/eaaccessorymanager/1613913-showbluetoothaccessorypicker. So you can do this in-app, but not with app UI.

2. Connecting to a BTc most likely means connecting RFCOMM on a certain UUID. 
For Android, there is a well-known UUID (https://stackoverflow.com/a/21760106/375209) which connects to RFCOMM serial port. Some devices may support additional UUIDs (which should be discovered). This used to be integer based ports, but the integer based API is hidden. 

On iOS, you'd open a `EASession` to a `EAAccessory` as discovered earlier: https://developer.apple.com/documentation/externalaccessory/easession

On either platform, once connected, you'll get access to an input & output stream for transfer.

@pauldemarco do you think this is something a `flutter_classic_blue` plugin would be more suitable? It seems like a different architecture than what you have implemented now.

pauldemarco, 2019-02-02:
I would like to entertain the possibility of including Classic into this package.  The API would need to be refactored to include that functionality onto the BluetoothDevice class --much like Android has it.  

But contrastingly, if the developer is for the most part only ever going to use BLE or Classic, it might make more sense to keep them separate.

I'm leaning towards 2 packages for now... both adapting until they have cleanest/most robust API possible, then reconsider merging together to have an all-in-one solution to Flutter Bluetooth.

GPlay97, 2019-02-07:
@pauldemarco Would be amazing!
I'm looking for something like https://github.com/don/BluetoothSerial

Missing Bluetooth Classic support in Flutter is currently preventing me from giving it a try

MuhammadElhrz, 2019-03-03:
@pauldemarco I wonder how there is no single useful library to connect with classical bluetooth devices in flutter?!
That is very important thing to have.
In my case i want to connect the app i want to make with a heart rate sensor through bluetooth and that sensor is not a BLE device, how could i do that know?

tresheffron, 2019-05-23:
I also have a use case for connecting to a heart rate sensor that isn't currently identified by flutter_blue during a scan.

kenreilly, 2019-05-31:
I wish I had known this earlier today before spending hours implementing the wrong solution :(

PsychoXIVI, 2019-07-02:
There is [`flutter_bluetooth_serial`](https://github.com/edufolly/flutter_bluetooth_serial/) for serial communication (classic) of Bluetooth.

aytunch, 2019-10-28:
@pauldemarco Hi Paul, may I ask if there is any plans for the Bluetooth Classic support? Thanks.

PsychoXIVI, 2019-10-28:
You should use https://github.com/edufolly/flutter_bluetooth_serial for Bluetooth Classic (Serial) I think.

aytunch, 2019-10-28:
> https://github.com/edufolly/flutter_bluetooth_serial

@PsychoXIVI Unfortunately there is support for iOS :(

PsychoXIVI, 2019-10-28:
@aytunch It's because iOS does not allow unauthorized (by Apple) apps to use raw Bluetooth sockets API as far I know.

ride4sun, 2020-04-09:
@PsychoXIVI that is correct but the API could probably be integrated into https://github.com/edufolly/flutter_bluetooth_serial to support IOS.

I think flutter_blue should stay just on BLE to keep the API simpler. 
Mixing use cases is not a good idea.