pauldemarco/flutter_blue #120: Flutter Blue scanResult.advertisementData.connectable returning "false"

weekaung, 2018-10-11:
Hi Paul,
The connectable field is returning "false" when deploying on a device with Android 5.0 but when deployed onto an Android 8.0 device, it is returning "true", which is the correct status. I have no problems connecting to the bluetooth device using native Android code whether on Android 5.0 or 8.0.
So why is "flutter blue" returning "false", when is should be "true".
Thanks in advance,
David.

Fleximex, 2018-10-24:
For Bluetooth you need permissions. But up until Android 5 those permission were asked when installing the app. After Android 5 permission are asked in the app when needed.
So maybe the Android 5 phone has never given the app permission to use Bluetooth (or location (needed for Bluetooth)).

GauravPatni, 2019-02-08:
Hi weekaung ,

I am facing same problem .. please help if you have resolved issue ....

Fleximex, 2019-02-08:
Though probably not the problem, but just to be sure: for (most) non BLE devices that Flutter Blue finds, returning false is normal behaviour (https://github.com/pauldemarco/flutter_blue/issues/125#issuecomment-432684180).

GauravPatni, 2019-02-09:
for more information , please refer following issue [#186](https://github.com/pauldemarco/flutter_blue/issues/186)

GauravPatni, 2019-02-13:
Hi weekaung ,

Have you solved 'The connectable field is returning "false" '  issue ?