pauldemarco/flutter_blue #125: Can' get the example installed

DerMarco, 2018-10-21:
Hello guys. 

So far I've got zero experience with flutter and only few with github and android studio.

I was trying to import the project and get the example to run on my phone but I always get errors of different kinds.

Could someone give me a quick overview of what to pay attention to?

I may as well just forget something while getting the file from GitHub and importing this to android studio...

Fleximex, 2018-10-24:
The example is somewhat embedded in the whole Flutter Blue library. For example it refers to the Flutter Blue package as a relative path: `..\`

An easier method is to just start a new Flutter Blue project. Delete `main.dart` in the lib folder, of your newly created project, and delete the test folder.

Download https://github.com/pauldemarco/flutter_blue/archive/master.zip
After that copy the files from /example/lib/ out of the downloaded zip into the lib folder of your project in file explorer.

Those files will give error because you need to add Flutter Blue to your project. In Android Studio open `pubspec.yaml` and add the line `flutter_blue: ^0.4.1` underneath `cupertino_icons: ^0.1.2`. Then click on 'Packages get' in the Flutter command bar (which appear when you have pubspec.yaml open).

If you run the project now you will get an error Flutter Blue requires minimum SDK version 19 (Android 4.4) but the default is minimum is set to 16 (Android 4.1). Go to the android folder, then app folder and open build.gradle. and change the line `minSdkVersion 16` to `minSdkVersion 19`.

Run the app using a phone. In Android Studio you can't simulate Bluetooth so you need a real phone. On the It will run the app. The step (shown in the console) Gradle task 'assembleDebug'... may take a while so be patient. The first time your run you might see some red warnings in the console.

If all works then the app will ask permission to use location and work.

I used Flutter 0.10.1 and Flutter Blue 0.4.1 for this.

DerMarco, 2018-10-24:
Thanks a lot for this detailed Explanation!

I am currently having Issues running this on my phone (Wiko) , but I will have Access to a Nexus Tablet in a bit.

Will try to run it then and if it works, close the issue.

DerMarco, 2018-10-24:
So i've actually been able to get the example app to run on a Nexus 7 Tablet with Android 4.3.

So far after starting I can click the floating button to open a list of bluetooth devices.

The only problem here is that I don't get any readable device names displayed, only hex numbers instead.

Furthermore the "connect" button is greyed out and clicking it only results in expanding the tile.

As far as I've read through the code there has to be more functionality to it than the one I have availible so far. Am I right?

And what could cause the issue? I'm trying out myself as well...

Fleximex, 2018-10-24:
Simple answer is you are probably not finding any BLE (Bluetooth Low Energy) devices.

I'm relatively new to Bluetooth and Flutter myself. But my guess is those seemingly random numbers you see are non BLE (a.k.a. Bluetooth Classic) devices or just mac adresses of wifi enabled devices since Bluetooth shares the same 2.4GHz band.

An easy fix I use is to check for a name. So in the method `_startScan()` in `main.dart` in the example app you can replace the code
setState(() {
   scanResults[scanResult.device.id] = scanResult;
});
with
```
if (scanResult.advertisementData.localName.length > 0)
{
   setState(() {
      scanResults[scanResult.device.id] = scanResult;
   });
}
```

Now it will show only BLE devices (from which it can obtain a name).

DerMarco, 2018-10-24:
Okay so this means that I am only able to connect to BLE devices, and if so: how?

The goal of my project is to receive data from an arduino board and as a first step receive and display a string sent from another phone using serial bluetooth terminal app, just to get started. Would this be possible with FlutterBlue? :)

DerMarco, 2018-10-25:
Another Question: Does Flutter Blue only support Bluetooth Low Engery or the "normal" one as well?

Concretely I Need to connect to a HC-06 Arduino module, and at the Moment I'm not sure if this is actually possible ...

Fleximex, 2018-10-25:
I don't think it's possible because that module uses Bluetooth 2.0 and only supports Bluetooth classic.

You would need the HC-10 which uses Bluetooth 4.0 and thus supports both classic and BLE.

Bluetooth classic is still an open issue:
https://github.com/pauldemarco/flutter_blue/issues/118

jonhunterking, 2019-11-13:
> The example is somewhat embedded in the whole Flutter Blue library. I.e. it refers to the Flutter Blue package as a relative path: `..\`
> 
> An easier method is to just start a new Flutter Blue project. Delete `main.dart` in the lib folder, of your newly created project, and delete the test folder.
> 
> Download https://github.com/pauldemarco/flutter_blue/archive/master.zip
> After that copy the files (`main.dart` and `widgets.dart`) from /example/lib/ out of the downloaded zip into the lib folder of your project in file explorer.
> 
> Those files will give an error because you need to add Flutter Blue to your project. In Android Studio open `pubspec.yaml` and add the line `flutter_blue: ^0.4.1` underneath `cupertino_icons: ^0.1.2`. Then click on 'Packages get' in the Flutter commands bar (which appears when you have pubspec.yaml open).
> 
> If you run the project now you will get an error because Flutter Blue requires minimum SDK version 19 (Android 4.4) but the default minimum is set to 16 (Android 4.1). Go to the android folder in your project, then app folder and open build.gradle. and change the line `minSdkVersion 16` to `minSdkVersion 19`.
> 
> Run the app using a phone. In Android Studio you can't simulate Bluetooth so you need a real phone. Now run the app. The step (shown in the console) `Gradle task 'assembleDebug'...` may take a while, so be patient. The first time you run you might see some red warnings in the console but you can ignore those if the app starts normally.
> 
> If all works then the app will ask permission to use location (needed for Bluetooth) and you can scan for Bluetooth devices using the search button.
> 
> I used Flutter 0.10.1 and Flutter Blue 0.4.1 for this.

When I try and follow these instructions, after I get packages I still have an error in my main.dart that "import 'package:flutter_blue_example/widgets.dart';" target of URI does not exist. 

Any idea how to fix that?