pauldemarco/flutter_blue #127: I can write a characteristic with Android but not with iOS

lemonbuzz, 2018-10-22:
So I'm developing a BLE device with an ESP32 using Arduino core library. To control this device with a smartphone, I decided to use Flutter and this very nice library. By the way thank you for the time you took to create it!

I've been facing an issue though and I'm not really sure where to start in order to try to fix it.

To put it simply, I can successfully connect to the device on both platforms. However, when I try to write to a characteristic on iOS, nothing happens and my device doesn't receive anything. Everything works fine with Android and my device gets written succesfully.

I'm not very experienced with BLE and I simply took an example code for the BLE device side of things...Maybe it's not configured correctly and doesn't work the way BLE is implemented on iOS? Eitherway I'm lost and not sure where to start!

I welcome any help :)

lemonbuzz, 2018-10-24:
Out of no where, it started working...Mysteriously

closed 2018-10-24 by lemonbuzz

aravindara, 2018-11-09:
For me it is still not working

lemonbuzz, 2018-11-09:
Honestly, all I did ( which i'm pretty sure had no actual effects, just a coincidence ) was to add the await keyword in front of the BluetoothDevice.writeCharacteristic call and include the optional 'type' parameter to the method.

I went from this:
`BluetoothDevice.writeCharacteristic(BluetoothCharacteristic, [0x12, 0x30])`

To this:
`await BluetoothDevice.writeCharacteristic(BluetoothCharacteristic, [0x12, 0x30], type: CharacteristicWriteType.withResponse)`

neerajyadv, 2019-03-30:
> I went from this:
> `BluetoothDevice.writeCharacteristic(BluetoothCharacteristic, [0x12, 0x30])`
> 
> To this:
> `await BluetoothDevice.writeCharacteristic(BluetoothCharacteristic, [0x12, 0x30], type: CharacteristicWriteType.withResponse)`
> 
> And it started working on iOS...I haven't tried to play around with it to see if it actually did anything, just reporting you what i've experienced!

It is not working on ios even after this code change.