pauldemarco/flutter_blue #129: Feature request: Request MTU size

RyanYANG52, 2018-11-01:
First of all, thanks for this awesome project.

I'm write an app using BLE to listen to some sensor data, I was able to receive notify value on a characteristic. But mtu size is 23 bytes by default, so I can only read the first 20 bytes, I didn't find any api to request a large size. My sensor data length is mutable and could be large than 23.

My app is running on > Android API 21. Here I found a [stackoverflow post](https://stackoverflow.com/questions/30904341/requesting-mtu-with-bluetooth-low-energy-connection-on-android-4-3-4-4-api-18-2)

bondjpf, 2019-01-09:
MTU size is very important. I hope devs can add this feature so more people can start using this.

wabash9000, 2019-01-22:
+1 For this.  On iOS I don't have to do this and can very quickly send 136 byte packets.  On Android I am unable to send anything greater than 23 bytes and the transfer takes quite a bit longer because of the overhead built into BLE.

I found that I can receive 136 byte packets from the bluetooth device, but I am unable to send them.  If I send an 8 byte packet on the characteristic, the device will reply with a 136 byte packet.

RyanYANG52, 2019-01-25:
you can create a pull request @wabash9000

LastMonopoly, 2019-04-28:
If you want to send more than 23 bytes on android, use writeWithResponse instead of writeWithoutResponse

pauldemarco, 2019-09-11:
@RyanYANG52  @bondjpf @LastMonopoly 
I have implemented MTU functionality in the latest version of the plugin (0.6.1).
Please try it out and let me know your feedback.

closed 2019-09-11 by pauldemarco