pauldemarco/flutter_blue #130: Cannot run flutter blue example on iPhone

rschlund, 2018-11-04:
When i try to deploy the flutter blue example to an iPhone, i get the error: 
Code Signing Error: No profiles for 'com.pauldemarco.flutterBlueExample' were found and I'm also unable to create a provisioning profile in Xcode with the same reason. Is there a way to avoid having to provide a provisioning profile for Paul Demarco?

cesear, 2018-12-05:
This is an issue related to your apple developer account and have nothing to do with the example project as it works for me just fine in iOS device, if you already have an apple developer certificate  open the iOS module with xcode and go under PROJECT, then select Runner under Targets and change the team to your team make sure to select Automatically manage signing, else go and create a developer certificate and add it to xcode. I hope this will help you

pauldemarco, 2019-06-18:
@rschlund as @cesear, please make sure your xcode is properly setup with your developer account.

Closing this issue for now.

closed 2019-06-18 by pauldemarco