pauldemarco/flutter_blue #131: Unable to write a characteristics in IPhone

aravindara, 2018-11-09:
Same app which is running perfectly in Android is failed to write characteristics in IOS

pauldemarco, 2018-12-07:
@aravindara I would need some more information if I'm to try and solve the issue.

What does the code look like?

What is the structure of the Bluetooth Characteristic / does it have descriptors?

aduarte88, 2019-01-25:
I have the same problem
in android works but in ios not
I have already been able to write in iOS of type "withoutResponse" but when it is "withResponse" it gives this error

tgeukens, 2019-01-28:
@aduarte88  I had a similar issue running the basic example in the project.  Double check whether the characteristic actually has the 'write' flag set in the characteristic declaration.  On iOS you'll indeed get an error trying to write with response on a characteristic that only supports write without response.  You can double check the characteristic's declaration by inspecting the characteristic.properties.writeWithoutResponse and characteristic.properties.write flags.  The example does not differentiate between these 2 types of writes (yet).  I'm pretty new to dart/flutter but this should fix the example I believe:

```
_writeCharacteristic(BluetoothCharacteristic c) async {
    CharacteristicWriteType writeType = c.properties.writeWithoutResponse
        ? CharacteristicWriteType.withoutResponse
        : CharacteristicWriteType.withResponse;
    await device.writeCharacteristic(c, [0x12, 0x34], type: writeType);
    setState(() {});
  }```

100cm, 2019-01-28:
i have fixed this by listen the notification.

pauldemarco, 2019-06-18:
This issue seems to be resolved. Closing for now.

closed 2019-06-18 by pauldemarco

micromotores, 2021-02-22:
i have a this problem ,help me please
[image]