pauldemarco/flutter_blue #133: Upgrade android protobuf

csga5000, 2018-11-12:
With the newest version of flutter I had a couple problems running my project with flutter blue.

First of all, there was an error saying your version of protobuf was lower than supported.
I fixed this by changing android/build.grade in your plugin line 12:
```
        classpath 'com.google.protobuf:protobuf-gradle-plugin:0.8.7'
```

Then I got an error saying my minSdk version was lower than yours, so I changed mine from 16 to 19.

Lastly I got these warnings:
```
Note: C:\Users\csga5\AppData\Roaming\Pub\Cache\hosted\pub.dartlang.org\flutter_blue-0.4.1\android\src\main\java\com\pauldemarco\flutterblue\FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\dev\goplug\mobile\build\flutter_blue\generated\source\proto\debug\javalite\com\pauldemarco\flutterblue\Protos.java uses unchecked or unsafe operations.
```

Some of these issues may be antipicated, but at the very least, I would assume there is no harm in you upgrading the version of protobuf you're requesting in your build.gradle.  Ideally the other hiccups wouldn't happen or are well documented

jairwen, 2018-11-18:
same here, and only work on ios.

CTaszus, 2018-11-20:
Same problems here. Fix would be great.

csga5000, 2018-11-20:
Created a pull request.  In the mean time if @CTaszus or others are having this issue, you can reference my cloned version of the repo: https://github.com/csga5000/flutter_blue

Or look at .flutter-plugins, find the directory for flutter-blue and modify android/build.gradle within that folder to have an updated protobuf. (By modifying to: classpath 'com.google.protobuf:protobuf-gradle-plugin:0.8.7')

CTaszus, 2018-11-20:
Thanks for your offer @csga5000 - I modified the files as you wrote in your first post.

csga5000, 2018-11-20:
Okay, I mentioned the 3 problems I had getting setup.

1. Protobuf version
2. Default minSdk needed updating
3. Get warnings when running

1. Should be addressed by: https://github.com/pauldemarco/flutter_blue/pull/135
2. Should be addressed by better documentation: https://github.com/pauldemarco/flutter_blue/pull/136
3. Not certain why those are occurring.  It looks like your code appropriately handles this issue: https://stackoverflow.com/questions/30223071/startlescan-replacement-to-current-api, so I'm not sure what other issues there may be and don't have time to actually debug much further.  It seems like it would be something trivial.  Either way, it's not as probablematic as 1 or 2.

pauldemarco, 2018-12-22:
Fixed in 2cf8cecb2f0f2bd80dd8562fa9ed8de0102954ef

Opened a new issue regarding the warning message when compiling for Android.

closed 2018-12-22 by pauldemarco