pauldemarco/flutter_blue #138: writeCharacteristic() unreliable

lcrocker, 2018-12-06:
Running up-to-the-minute flutter SDK (2018-12-06), target is Moto G (3rd generation) running Android 6.0. Bluetooth object finds my device, reads the services (the one I'm using is Nordic Semiconductor's "NUS" service, which is kind of a terminal emulator: it allows sending arbitrary packets of text back and forth). I setNotify on the _rChar characteristic and I'm getting those. And I write to the _wChar characteristic with this function:

```
  Future<void> write(String s) async {
		print('BT65: ${s.codeUnits}');
		if (null == _wChar) { return; }

		await _device.writeCharacteristic(_wChar, s.codeUnits,
			type: CharacteristicWriteType.withoutResponse ).catchError((err) {
			print('BT67: Error: $err');
		});
	}
```
It never prints the error, but it succeeds only about half the time. Log fragment below. When it succeeds, I see an "onCharacteristicChanged", and my device responds with the text "Count: XXXX" (first two sends below work fine). But about half the time, the send never arrives, but no error is triggered, I just get no response (second two attempts, for example). The devices are right next to each other, and I'm sending on a button press, so radio strength or speed should not be an issue).

I/flutter (32629): BT104: wChar: Instance of 'BluetoothCharacteristic'
I/flutter (32629): BT65: [65, 98, 67, 100, 69, 102]
D/FlutterBluePlugin(32629): onCharacteristicWrite: 
D/FlutterBluePlugin(32629): onCharacteristicChanged: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
I/flutter (32629): BT115: Text: Count: 406365
I/flutter (32629): BT65: [65, 98, 67, 100, 69, 102]
D/FlutterBluePlugin(32629): onCharacteristicWrite: 
D/FlutterBluePlugin(32629): onCharacteristicChanged: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
I/flutter (32629): BT115: Text: Count: 408679
I/flutter (32629): BT65: [65, 98, 67, 100, 69, 102]
D/FlutterBluePlugin(32629): onCharacteristicWrite: 
I/flutter (32629): BT65: [65, 98, 67, 100, 69, 102]
D/FlutterBluePlugin(32629): onCharacteristicWrite: 
I/flutter (32629): BT65: [65, 98, 67, 100, 69, 102]
D/FlutterBluePlugin(32629): onCharacteristicWrite: 
D/FlutterBluePlugin(32629): onCharacteristicChanged: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
I/flutter (32629): BT115: Text: Count: 410867
I/flutter (32629): BT65: [65, 98, 67, 100, 69, 102]
D/FlutterBluePlugin(32629): onCharacteristicWrite: 
I/flutter (32629): BT65: [65, 98, 67, 100, 69, 102]
D/FlutterBluePlugin(32629): onCharacteristicWrite: 
I/flutter (32629): BT65: [65, 98, 67, 100, 69, 102]
D/FlutterBluePlugin(32629): onCharacteristicWrite: 
D/FlutterBluePlugin(32629): onCharacteristicChanged: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
I/flutter (32629): BT115: Text: Count: 419323
Application finished.

pauldemarco, 2018-12-06:
Hi Lee Daniel Crocker,

What does the calling code look like?  Are you also "awaiting" the call to
the write function?

Warm Regards,
Paul

lcrocker, 2018-12-06:
It's being called from a non-async function, so I was just calling it as a future; I added the then() clause (I already had the catchError()), and it is always triggering, both on success and failure, so this code always ends up printing "Done.".

```
  void connectButtonAction(context) {
    if (5 == _connectState) {
        print('CN56: Sending...');
        app.ble.write('AbCdEf').then((r) {
          print('CN57: Done.');
        }).catchError((err) {
          print('CN58: Error: $err');
        });
        return;
    }
```

lcrocker, 2018-12-07:
I refactored the code to make it much simpler. Now the functions are:
```
Future<void> sendPress() async {
    await app.ble.send('AbCdEf');
}
```
and
```
Future<void> send(String s) async {
    assert(null != _wChar);
    await _device.writeCharacteristic(_wChar, s.codeUnits);
}
```
...and I get the same result. First two or three work, then it's about 50/50.

pauldemarco, 2018-12-07:
@lcrocker How often is this code being called?

lcrocker, 2018-12-07:
Right now it's just when I press a button, every few seconds. Eventually I'll need it to send commands to the device pretty regularly, many packets per second. I've done this with Cordova and Noble and gotten roughly 1k per second throughput sending a few kb without error (it does tend to drop out every few kb, so my protocol sometimes has to restart, but the basic functionality works).

lcrocker, 2018-12-07:
Don't see much difference in the ultimate code between you and cordova-plugin-ble-central. It all comes down to a call to BluetoothGatt.writeCharacteristic(), which is synchronous and returns true on success. The only major difference in the structure of your code and his is that he queues all the commands and services them from the queue rather than executing them on demand. But if there's something about the context that would cause the call to fail, you'd still expect it to actually fail and return false. Curious.

lcrocker, 2018-12-11:
FYI: Changed the firmware on my remote device, and the failure rate went down from 50% to about 2%, so you may be off the hook on this one. Feel free to close or de-prioritize unless you find something else.

pauldemarco, 2018-12-11:
@lcrocker I haven't responded yet since I'm in the process of creating something very similar, at much higher bandwidth.  I will close for now, and re-open if I run into issues.

Feel free to do the same.

Thanks.

closed 2018-12-11 by pauldemarco