pauldemarco/flutter_blue #140: Use in isolate?

lcrocker, 2018-12-08:
Has anyone ever used this library in an isolate? Seems like a natural thing to me, but I just refactored my code to do that and I'm getting an error from "state". This simple startup code;

```
    FlutterBlue _ble = FlutterBlue.instance;
    _ble.state.then((s) {
      print('FB: $s');
    }).catchError((err) {
      print('FB: $err');
    });
```
prints "FB: error: native function 'Window_sendPlatformMessage' (4 arguments) cannot be found"

sendPlatformMessage is part of Flutter's GUI code, and doesn't appear in your code anywhere. I'll try to see if I can make a minimal example for you.

pauldemarco, 2018-12-11:
@lcrocker Platform messages are for communication between Dart <-> Android/iOS.

It's been awhile since I've updated myself on the work being done to support background isolates, and I'm not sure if it will require changes to the plugin.

I'll subscribe to flutter/flutter#13937 and keep updated on it.

rmawatson, 2019-02-14:
https://github.com/pauldemarco/flutter_blue/pull/191 should let you run in an isolate.

MichealReed, 2019-08-06:
@rmawatson been at this all day. I pulled #191 directly and also merged with master to gain access to the new api. Everything compiles and runs, but I cannot get it to scan properly. 

Here is my debug output:

[image]

Kushal5, 2020-03-24:
Is there any update? Do Isolates work with this?

marcmathys, 2020-06-14:
I have the same question.

JuniorJPDJ, 2021-01-16:
+