pauldemarco/flutter_blue #141: Protobuf Gradle plugin version 0.8.6 and higher vs Project 'flutter_blue' is using version 0.8.3

icarus31, 2018-12-10:
Using:
- VS Code: 1.29.1
- Flutter: 1.0.0
- Flutter_blue: 0.4.1

When starting the debugger, I got:

```
* Error running Gradle:
ProcessException: Process "C:\Development\flutter\bluetooth_example\android\gradlew.bat" exited abnormally:
> Configure project :flutter_blue
The com.google.protobuf plugin was already applied to the project: :flutter_blue and will not be applied again after plugin: android-library

FAILURE: Build failed with an exception.
* What went wrong:
The Android Gradle plugin supports only Protobuf Gradle plugin version 0.8.6 and higher. Project 'flutter_blue' is using version 0.8.3.
```

pauldemarco, 2018-12-22:
Fixed in 2cf8cecb2f0f2bd80dd8562fa9ed8de0102954ef

closed 2018-12-22 by pauldemarco

darnoldtux, 2018-12-28:
I am having the same problem as @ai-curry. I have not been able to find a work around.

pauldemarco, 2018-12-28:
This is fixed on master branch. I will publish a patch on publang today.

Thanks for your patience.