pauldemarco/flutter_blue #153: My phone (Meizu 7 Pro) can't find any device using flutter_blue

woozbooz, 2018-12-18:
Hi. I'm working on some project and i try to use flutter_blue plugin.
Unfortunattely your sample app cant find any active device around.
Is this only problem with android or with my phone?
Btw built-in bluetooth tool can find other devices without problem.

woozbooz, 2018-12-19:
Also try it on other devices:
iPad FUN 64
iOS 11.4.1

iPhone 5c
iOS 10.3.3

Prestigio Wize 3147
Android 6.0

Result is the same - no devices found.

weicheng2138, 2018-12-19:
Me too
Same problem just like you. Cannot scan out any devices, although the scan process bar works so fine.
But I'm curious about the readme below
> Scanning for service UUID's doesn't return any results

Is there any other settings that I missed? @pauldemarco

pauldemarco, 2018-12-19:
Hello,

Are these devices advertising a Bluetooth Low Energy service?

FlutterBlue does not support classic Bluetooth at the present time.

ride4sun, 2018-12-21:
@weicheng2138  could you please be so kind to point me to the bluetooth spec you are talking about.

dieudv, 2019-01-30:
Me too. I'm try it on:

Samsung Galaxy Note 3 SM-N900 (android 5.0)
Samsung Galaxy SM-J500H (android 6.0.1)
Redmi Note 3 (android 6.0.1)

All my devices turn on bluetooth but they can't find each other.

LaxmikanthMadhyastha, 2019-06-28:
any update on this?? I'm facing the same issue

luissouza, 2019-08-08:
any update on this?? I'm facing the same issue!!

LaxmikanthMadhyastha, 2019-08-09:
@luissouza  as far as I know you can't really find mobile devices, it will easily discover light bluetooth devices like a band and all

johngorithm, 2020-04-29:
> Hello, Are these devices advertising a Bluetooth Low Energy service? FlutterBlue does not support classic Bluetooth at the present time.

When should we expect support for Classic Bluetooth devices? Thanks.